ucb-bar/chisel-tutorial #42: README

schoeberl, 2016-06-08:
Wouldn't it be nicer and shorter to show the make target for the examples, such as "make Mux4.out" in folder problem instead of the verbose sbt string? The complete sbt string could be shown for a singe example.

If you agree on this simplification, I can do the change with a pull request.

Martin

chick, 2016-06-09:
We have been moving away from make, particularly in the tutorial, towards sbt for it's dependency management.  The README does reference specific sbt run commands, but it does reference the run scripts provided. ```./run-problems.sh all``` can also be used as ```./run-problems.sh Mux2``` to run specific examples.  All this is on the reorganize branch, which should be merged with master shortly. Is this where you have been testing?

schoeberl, 2016-07-18:
No I was testing in the master branch, just as a preparation for a chisel course I gave. If this is reorganized with sbt and short calls to shell scripts it is fine. Sorry for the very late response. Probably you could close this issue then.

ucbjrl, 2016-10-13:
Closed due to new file structure.

closed 2016-10-13 by ucbjrl