ucb-bar/chisel-tutorial #46: Terminal initialization failed, JLine incompatibility detected

awmiller, 2016-09-11:
**Using the Vagrant tutorial**

vagrant@precise64:~/chisel-tutorial/hello$ make
set -e -o pipefail; "sbt" -Dsbt.log.noformat=true -DchiselVersion="2.+" "run Hello --genHarness --compile --test --backe
nd c --vcd --targetDir /home/vagrant/chisel-tutorial/generated/hello " | tee /home/vagrant/chisel-tutorial/generated/hel
lo/Hello.out
tee: /home/vagrant/chisel-tutorial/generated/hello/Hello.out: No such file or directory
[ERROR] Terminal initialization failed; falling back to unsupported
java.lang.IncompatibleClassChangeError: Found class jline.Terminal, but interface was expected
        at jline.TerminalFactory.create(TerminalFactory.java:101)
        at jline.TerminalFactory.get(TerminalFactory.java:159)
        at sbt.ConsoleLogger$.ansiSupported(ConsoleLogger.scala:123)
        at sbt.ConsoleLogger$.<init>(ConsoleLogger.scala:117)
        at sbt.ConsoleLogger$.<clinit>(ConsoleLogger.scala)
        at sbt.GlobalLogging$.initial(GlobalLogging.scala:43)
        at sbt.StandardMain$.initialGlobalLogging(Main.scala:60)
        at sbt.StandardMain$.initialState(Main.scala:69)
        at sbt.xMain.run(Main.scala:28)
        at xsbt.boot.Launch$.run(Launch.scala:55)
        at xsbt.boot.Launch$$anonfun$explicit$1.apply(Launch.scala:45)
        at xsbt.boot.Launch$.launch(Launch.scala:60)
        at xsbt.boot.Launch$.apply(Launch.scala:16)
        at xsbt.boot.Boot$.runImpl(Boot.scala:31)
        at xsbt.boot.Boot$.main(Boot.scala:20)
        at xsbt.boot.Boot.main(Boot.scala)

java.lang.IncompatibleClassChangeError: JLine incompatibility detected.  Check that the sbt launcher is version 0.13.x o
r later.
        at sbt.ConsoleLogger$.ansiSupported(ConsoleLogger.scala:134)
        at sbt.ConsoleLogger$.<init>(ConsoleLogger.scala:117)
        at sbt.ConsoleLogger$.<clinit>(ConsoleLogger.scala)
        at sbt.GlobalLogging$.initial(GlobalLogging.scala:43)
        at sbt.StandardMain$.initialGlobalLogging(Main.scala:60)
        at sbt.StandardMain$.initialState(Main.scala:69)
        at sbt.xMain.run(Main.scala:28)
        at xsbt.boot.Launch$.run(Launch.scala:55)
        at xsbt.boot.Launch$$anonfun$explicit$1.apply(Launch.scala:45)
        at xsbt.boot.Launch$.launch(Launch.scala:60)
        at xsbt.boot.Launch$.apply(Launch.scala:16)
        at xsbt.boot.Boot$.runImpl(Boot.scala:31)
        at xsbt.boot.Boot$.main(Boot.scala:20)
        at xsbt.boot.Boot.main(Boot.scala)
Error during sbt execution: java.lang.IncompatibleClassChangeError: JLine incompatibility detected.  Check that the sbt
launcher is version 0.13.x or later.
make: *** [/home/vagrant/chisel-tutorial/generated/hello/Hello.out] Error 1
vagrant@precise64:~/chisel-tutorial/hello$

ucbjrl, 2016-09-12:
> at sbt.ConsoleLogger$.(ConsoleLogger.scala:117)
> at sbt.ConsoleLogger$.(ConsoleLogger.scala)

Thanks for pointing this out. We'll upgrade this this week.

ucbjrl, 2016-09-19:
I've upgraded the Vagrant image with newer versions of ubuntu, java, and sbt. The Chisel code is very old.

We're on the verge of releasing updates to Chisel2 and a SNAPSHOT release of Chisel3. Check the [web site](https://chisel.eecs.berkeley.edu) for announcements.