ucb-bar/chisel-tutorial #53: dependency not found

colin4124, 2016-12-12:
I used master branch, 8b306ae3b24a4daec55e084072518eee3504154e

```
 edu.berkeley.cs:chisel-iotesters_2.11:1.2-SNAPSHOT:
    not found:
      /home/colin4124/.ivy2/local/edu.berkeley.cs/chisel-iotesters_2.11/1.2-SNAPSHOT/ivys/ivy.xml
      https://repo1.maven.org/maven2/edu/berkeley/cs/chisel-iotesters_2.11/1.2-SNAPSHOT/chisel-iotesters_2.11-1.2-SNAPSHOT.pom
      https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel-iotesters_2.11/1.2-SNAPSHOT/chisel-iotesters_2.11-1.2-SNAPSHOT.pom
      https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel-iotesters_2.11/1.2-SNAPSHOT/chisel-iotesters_2.11-1.2-SNAPSHOT.pom
  edu.berkeley.cs:chisel3_2.11:3.1-SNAPSHOT:
    not found:
      /home/colin4124/.ivy2/local/edu.berkeley.cs/chisel3_2.11/3.1-SNAPSHOT/ivys/ivy.xml
      https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.11/3.1-SNAPSHOT/chisel3_2.11-3.1-SNAPSHOT.pom
      https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.11/3.1-SNAPSHOT/chisel3_2.11-3.1-SNAPSHOT.pom
      https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel3_2.11/3.1-SNAPSHOT/chisel3_2.11-3.1-SNAPSHOT.pom
```

I used release version, it does not support `val in  = Input(UInt(1.W)) :value W is not a member of Int`. The chisel3 version in build.sbt is 3.0-SNAPSHOT. Does 3.0-SNAPSHOT not support` UInt(1.W)` ?

So I want to try master which chisel3 version is 3.1-SNAPSHOT, but I encounter above problem and I can't test for it.

edwardcwang, 2016-12-12:
It appears that the .W notation was only added recently in b0cc0c93a80aec5bed54cfb11923636c09b7e180 so it doesn't exist in 3.0-SNAPSHOT, I think.

colin4124, 2016-12-12:
@edwardcwang Thinks very much.

I checkout master branch, and then `sbt run`. I encounter problem that "edu.berkeley.cs:chisel-iotesters_2.11:1.2-SNAPSHOT and edu.berkeley.cs:chisel3_2.11:3.1-SNAPSHOT not found". Do you encounter these problem, too ?

edwardcwang, 2016-12-12:
I don't believe that 3.1-SNAPSHOT of chisel3 has been released yet, so you'd have to either compile everything from source, or wait for a release.

colin4124, 2016-12-12:
@edwardcwang So that explains it. Thinks very much.

colin4124, 2016-12-12:
@edwardcwang I have done `sbt publish-local`, and there is `~/.ivy2/local/edu.berkeley.cs/chisel3_2.11/3.1-SNAPSHOT ` .  I just run `./run-problem.sh Adder` in master branch. It does not find locally but to  **https://oss.sonatype.org**. 

What should change on build.sbt ?

jackkoenig, 2016-12-12:
We are working on publishing an updated snapshot that includes recent API updates. Stay tuned.

In general, to run ahead of the published snapshot, publish-local is the right way to go, but you must publish-local other repos as well. https://github.com/ucb-bar/chisel-testers is the source of chisel-iotesters which relies on chisel3, firrtl (https://github.com/ucb-bar/firrtl) and the firrtl interpreter (https://github.com/ucb-bar/firrtl-interpreter). Also, be sure your build.sbt specifies using Scala 2.11.X since these repos are compiled for Scala 2.11, but by default, sbt uses 2.10.

colin4124, 2016-12-13:
@jackbackrack Thinks very much, it works!

colin4124, 2016-12-13:
@jackkoenig Thinks very much! It works!

closed 2016-12-13 by colin4124