ucb-bar/chisel-tutorial #55: sbt run target

bqwer, 2017-01-02:
I'm new to Scala and sbt and I can't figure out how `sbt run` target is specified.
I took interest in this tutorial structure as it can show how to make mixed projects with Scala and Chisel. And I couldn't get even first step. Why `sbt run` runs Hello.scala? I've red several sections of sbt documentation, searched StackOverflow and still I don't get how mainClass is specified to Hello.scala in chisel-tutorial. There is no mention of mainClass anywhere, no build.sbt rules, no Build.scala in project directory, there are other sources in src/main so probably Hello.scala isn't default. Where should I search for it?

jackkoenig, 2017-01-02:
This page has some useful information:
http://alvinalexander.com/scala/sbt-how-to-compile-run-package-scala-project

To summarize: `sbt` looks for `main` functions (or objects that extend `App` which in Scala makes an implicit `main` function) in `src/main/scala/...` recursively. Since there is only one (ie. `hello.Hello.main`) `sbt run` will run that one by default. If there were more than one, it running `sbt run` would prompt you to select which one it should run.

sbt also has supports organizing tests separately from your primary code (See http://www.scala-sbt.org/0.13/docs/Testing.html). It looks for test code in `src/test/scala/...` recursively. To run a main function in `test`, you run: `sbt test:run`. If you execute this command in this tutorial, you will see that there are multiple main functions defined in `test`.

Give it a try!

bqwer, 2017-01-02:
Thank you, now I got it! I've red article above and it mentions `main` method in one sentence. And sbt reference manual doesn't say a word about it: http://www.scala-sbt.org/0.13/docs/Running.html
So I was really confused and still find it somehow concealed.

jackkoenig, 2017-01-02:
Yeah I agree with you that it is weirdly not explained. I looked around briefly to point to find something that explained it but it wasn't super easy to find. Anyway, thanks for your question!

closed 2017-01-02 by jackkoenig