ucb-bar/chisel-tutorial #58: Cat(x,y) and Fill(x,y) are utilities and require separate import statement

bqwer, 2017-01-07:
To use `Cat(x,y)` and `Fill(x,y)` one need to `import chisel3.utils._` besides `import chisel3._`
It's demonstrated in examples so it's not critical issue but I believe it should be stated in Chisel Tutorial in explicit way.

When you are reading a tutorial you assume that you don't need to do anything special to be able to use all Chisel operators ( `Cat(x,y)` and `Fill(x,y)` are mentioned in section "5. Builtin Operators" and in "Table 1: Chisel operators on builtin data types" )

[Chisel Manual](https://chisel.eecs.berkeley.edu/2.2.0/chisel-manual.pdf) and [Chisel CheatSheet](https://chisel.eecs.berkeley.edu/2.2.0/chisel-cheatsheet.pdf) doesn't mention it either.
[Chisel 3 Wiki](https://github.com/ucb-bar/chisel3/wiki/Annotations%20Extending%20Chisel%20and%20Firrtl) mentions only `import chisel3._` in xviii and xix.

After reading Scala documentation I expected that `import chisel3._` would automatically import `chisel3.utils._` but it seems to be wrong. If you don't import `chisel3.utils._` while using `Cat(x,y)` or `Fill(x,y)` you will get compilation error: `not found: value Cat` or `not found: value Fill`

alaasal, 2019-11-02:
Hello, 
It is now ( chisel3.util._  ) there is no s in util.

younglimlim, 2022-04-04:
thanks, it fixes my problem.