ucb-bar/chisel-tutorial #66: discrepancy between Verilog and firrtl simulation

sols1, 2017-01-20:
The code passes the test:

```
class RealGCDInput extends Bundle {
  val a = UInt(16.W)
  val b = UInt(16.W)
}

class RealGCD extends Module {
  val io  = IO(new Bundle {
    val in  = DeqIO(new RealGCDInput())
    val out = Output(Valid(UInt(16.W)))
  })

  // Implement below ----------
  val valid = Reg(init = false.B)
  val done  = Reg(init = false.B)
  val a = Reg(UInt())
  val b = Reg(UInt())

  io.in.ready  := !valid
  io.out.valid := done
  io.out.bits  := a

  printf ("now  a=%d b=%d valid=%d io.in.valid=%d\n", a, b, valid, io.in.valid)
  when (!valid && io.in.valid ) {
    a     := io.in.bits.a
    b     := io.in.bits.b
    printf ("new  a=%d b=%d\n", a, b)
    valid := true.B
    done  := false.B
  }
  .elsewhen (valid) {
    when (a === b) {
      valid := false.B
      done  := true.B
      printf ("done a=%d b=%d\n", a, b)
    }
    .elsewhen (a < b) {
      b := b - a
      printf ("a<b  a=%d b=%d\n", a, b)
    }
    .otherwise {
      a := a - b
      printf ("b>a  a=%d b=%d\n", a, b)
    }
  }
  // Implement above ----------

}
```

However, the output does not have anything printed by `printf ("new  a=%d b=%d\n", a, b)`:

```
test:run-main problems.Launcher RealGCD
[info] Compiling 1 Scala source to /home/sols/src/chisel-tutorial/target/scala-2.11/classes...
[info] Running problems.Launcher RealGCD
Starting tutorial RealGCD
[info] [0.002] Elaborating design...
[info] [0.141] Done elaborating.
[info] [0.000] Elaborating design...
[info] [0.011] Done elaborating.
End of dependency graph
Circuit state created
SEED 1484943270543
now  a=48 b=32 valid=1 io.in.valid=1
b>a  a=48 b=32
now  a=16 b=32 valid=1 io.in.valid=0
a<b  a=16 b=32
now  a=16 b=16 valid=1 io.in.valid=0
done a=16 b=16
now  a=16 b=16 valid=0 io.in.valid=0
now  a=7 b=3 valid=1 io.in.valid=1
b>a  a=7 b=3
now  a=4 b=3 valid=1 io.in.valid=0
b>a  a=4 b=3
now  a=1 b=3 valid=1 io.in.valid=0
a<b  a=1 b=3
now  a=1 b=2 valid=1 io.in.valid=0
a<b  a=1 b=2
now  a=1 b=1 valid=1 io.in.valid=0
done a=1 b=1
now  a=1 b=1 valid=0 io.in.valid=0
now  a=100 b=10 valid=1 io.in.valid=1
b>a  a=100 b=10
now  a=90 b=10 valid=1 io.in.valid=0
b>a  a=90 b=10
now  a=80 b=10 valid=1 io.in.valid=0
b>a  a=80 b=10
now  a=70 b=10 valid=1 io.in.valid=0
b>a  a=70 b=10
now  a=60 b=10 valid=1 io.in.valid=0
b>a  a=60 b=10
now  a=50 b=10 valid=1 io.in.valid=0
b>a  a=50 b=10
now  a=40 b=10 valid=1 io.in.valid=0
b>a  a=40 b=10
now  a=30 b=10 valid=1 io.in.valid=0
b>a  a=30 b=10
now  a=20 b=10 valid=1 io.in.valid=0
b>a  a=20 b=10
now  a=10 b=10 valid=1 io.in.valid=0
done a=10 b=10
now  a=10 b=10 valid=0 io.in.valid=0
test RealGCD Success: 3 tests passed in 26 cycles taking 0.076821 seconds
RAN 21 CYCLES PASSED
Tutorials passing: 1
[success] Total time: 6 s, completed Jan 20, 2017 12:14:31 PM
```

jackkoenig, 2017-01-20:
It appears that the condition for executing `printf ("new  a=%d b=%d\n", a, b)` is simply never occurring. Looking at the `now ...` prints, the case `!valid && io.in.valid` never appears.

chick, 2017-01-20:
I can cause the *new* line to print with
```
  poke(c.io.in.bits.a, 4)
  poke(c.io.in.bits.b, 1)
  poke(c.io.in.valid, 1)

  step(1)
  peek(c.io.out.bits)
  poke(c.io.in.valid, 0)

  step(1)
  peek(c.io.out.bits)
  poke(c.io.in.valid, 0)

  step(1)
  peek(c.io.out.bits)
  poke(c.io.in.valid, 0)

  step(1)
  peek(c.io.out.bits)
  poke(c.io.in.valid, 1)

  step(1)
  peek(c.io.out.bits)
  poke(c.io.in.valid, 0)
```