ucb-bar/chisel-tutorial #81: sbt problem

kllbrd, 2017-03-27:
Hello,
I am facing an error while I'm trying to sbt run:
[error] (compile:compile) sbt.InvalidComponent: Could not find required component 'compiler-interface-src'
Any ideas anyone?
Thanks.

ucbjrl, 2017-03-27:
`compiler-interface-src` looks like name of an internal sbt library.

Could you show us the command you're using to run the tutorials and the complete output (including the above error)?

kllbrd, 2017-03-27:
Yes of course, I follow the instructions:
Getting the repo.
cd chisel-tutorial
sbt run
[root@localhost chisel-tutorial]# sbt run
[info] Loading project definition from /usr/local/chisel-tutorial/project
[info] Updating {file:/usr/local/chisel-tutorial/project/}chisel-tutorial-build...
[info] Resolving org.scala-sbt.ivy#ivy;2.3.0-sbt-2cc8d2761242b072cedb0a04cb39435[info] Resolving org.scala-lang#scala-reflect;2.10.4 ...
[info] Done updating.
[info] Set current project to chisel-tutorial (in build file:/usr/local/chisel-tutorial/)
[info] Updating {file:/usr/local/chisel-tutorial/}chisel-tutorial...
[info] Resolving jline#jline;2.12.1 ...
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.11/3.0-SNAPSHOT/chisel3_2.11-3.0-SNAPSHOT.jar ...
[info] 	[SUCCESSFUL ] edu.berkeley.cs#chisel3_2.11;3.0-SNAPSHOT!chisel3_2.11.jar (2917ms)
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/firrtl_2.11/1.0-SNAPSHOT/firrtl_2.11-1.0-SNAPSHOT.jar ...
[info] 	[SUCCESSFUL ] edu.berkeley.cs#firrtl_2.11;1.0-SNAPSHOT!firrtl_2.11.jar (3275ms)
[info] Done updating.
[info] Compiling 54 Scala sources to /usr/local/chisel-tutorial/target/scala-2.11/classes...
sbt.InvalidComponent: Could not find required component 'compiler-interface-src'
	at sbt.ComponentManager.invalid(ComponentManager.scala:57)
	at sbt.ComponentManager.sbt$ComponentManager$$notFound$1(ComponentManager.scala:34)
	at sbt.ComponentManager.sbt$ComponentManager$$createAndCache$1(ComponentManager.scala:37)
	at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:27)
	at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:26)
	at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:48)
	at xsbt.boot.Locks$.apply0(Locks.scala:38)
	at xsbt.boot.Locks$.apply(Locks.scala:28)
	at sbt.ComponentManager.lock(ComponentManager.scala:50)
	at sbt.ComponentManager.lockGlobalCache(ComponentManager.scala:49)
	at sbt.ComponentManager.sbt$ComponentManager$$fromGlobal$1(ComponentManager.scala:25)
	at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
	at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
	at sbt.ComponentManager.sbt$ComponentManager$$getOrElse$1(ComponentManager.scala:32)
	at sbt.ComponentManager$$anonfun$files$1.apply(ComponentManager.scala:44)
	at sbt.ComponentManager$$anonfun$files$1.apply(ComponentManager.scala:44)
	at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:48)
	at xsbt.boot.Locks$.apply0(Locks.scala:38)
	at xsbt.boot.Locks$.apply(Locks.scala:28)
	at sbt.ComponentManager.lock(ComponentManager.scala:50)
	at sbt.ComponentManager.lockLocalCache(ComponentManager.scala:47)
	at sbt.ComponentManager.files(ComponentManager.scala:44)
	at sbt.compiler.ComponentCompiler$$anonfun$compileAndInstall$1.apply(ComponentCompiler.scala:61)
	at sbt.compiler.ComponentCompiler$$anonfun$compileAndInstall$1.apply(ComponentCompiler.scala:58)
	at sbt.IO$.withTemporaryDirectory(IO.scala:285)
	at sbt.compiler.ComponentCompiler.compileAndInstall(ComponentCompiler.scala:58)
	at sbt.compiler.ComponentCompiler$$anonfun$getLocallyCompiled$1.apply$mcV$sp(ComponentCompiler.scala:47)
	at sbt.IfMissing$Define.apply(ComponentManager.scala:77)
	at sbt.ComponentManager.sbt$ComponentManager$$createAndCache$1(ComponentManager.scala:39)
	at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:27)
	at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:26)
	at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:98)
	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:81)
	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:102)
	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:102)
	at xsbt.boot.Using$.withResource(Using.scala:11)
	at xsbt.boot.Using$.apply(Using.scala:10)
	at xsbt.boot.Locks$GlobalLock.withFileLock(Locks.scala:102)
	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:62)
	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:52)
	at xsbt.boot.Locks$.apply0(Locks.scala:38)
	at xsbt.boot.Locks$.apply(Locks.scala:28)
	at sbt.ComponentManager.lock(ComponentManager.scala:50)
	at sbt.ComponentManager.lockGlobalCache(ComponentManager.scala:49)
	at sbt.ComponentManager.sbt$ComponentManager$$fromGlobal$1(ComponentManager.scala:25)
	at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
	at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
	at sbt.ComponentManager.sbt$ComponentManager$$getOrElse$1(ComponentManager.scala:32)
[... 44 lines cut ...]
	at sbt.compiler.AggressiveCompile.compile1(AggressiveCompile.scala:70)
	at sbt.compiler.AggressiveCompile.apply(AggressiveCompile.scala:45)
	at sbt.Compiler$.apply(Compiler.scala:70)
	at sbt.Defaults$.sbt$Defaults$$compileTaskImpl(Defaults.scala:736)
	at sbt.Defaults$$anonfun$compileTask$1.apply(Defaults.scala:730)
	at sbt.Defaults$$anonfun$compileTask$1.apply(Defaults.scala:730)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
[... cut ...]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[error] (compile:compile) sbt.InvalidComponent: Could not find required component 'compiler-interface-src'
[error] Total time: 13 s, completed Mar 27, 2017 10:29:38 PM

ucbjrl, 2017-03-27:
Thanks. I notice two issues:

- project/build.properties indicates we expect to build with sbt version 0.13.11, which in turn was built with Scala 2.10.6, but your sbt appears to require scala-reflect from 2.10.4
- the output is missing many expected `[info] Resolving` lines, which should be appearing prior to the `[info] downloading` line.

Googling this error indicates it may be due to problems with the sbt version or one of the ~/.sbt, ~/.m2, or ~/.ivy2 directories.

Could you provide information concerning your OS and sbt versions? For the latter, please send the output of the following:
```
cd chisel-tutorial
sbt about
```

kllbrd, 2017-03-27:
OS:
I am running Fedora 23 64-bit GNOME Version 3.18.2.
About sbt:
[root@localhost chisel-tutorial]# sbt about
[info] Loading project definition from /usr/local/chisel-tutorial/project
[info] Set current project to chisel-tutorial (in build file:/usr/local/chisel-tutorial/)
[info] This is sbt 0.13.11
[info] The current project is {file:/usr/local/chisel-tutorial/}chisel-tutorial 3.0-SNAPSHOT
[info] The current project is built against Scala 2.11.7
[info] 
[info] sbt, sbt plugins, and build definitions are using Scala 2.10.4

ucbjrl, 2017-03-27:
Are you running sbt as root? I would advise against this.
Apart from the security issues, sbt will need to write the ~/.sbt and ~/.ivy2 directories, and this can pose problems if these are accessed through a NFS mounted filesystem.

Can you run `sbt -d -v clean run` in the chisel-tutorial directory and send the output, please?

kllbrd, 2017-03-27:
By the way on my KDE Mint everything run perfectly.
Output of sbt -d -v clean without running it as root:

[kllbrd@localhost chisel-tutorial]$ sbt -d -v clean run
Getting org.scala-sbt sbt 0.13.11 ...
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/0.13.11/jars/sbt.jar ...
	[SUCCESSFUL ] org.scala-sbt#sbt;0.13.11!sbt.jar (3201ms)
downloading https://repo1.maven.org/maven2/org/scala-lang/scala-library/2.10.6/scala-library-2.10.6.jar ...
	[SUCCESSFUL ] org.scala-lang#scala-library;2.10.6!scala-library.jar (6770ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/main/0.13.11/jars/main.jar ...
	[SUCCESSFUL ] org.scala-sbt#main;0.13.11!main.jar (4742ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/compiler-interface/0.13.11/jars/compiler-interface.jar ...
	[SUCCESSFUL ] org.scala-sbt#compiler-interface;0.13.11!compiler-interface.jar (2454ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/actions/0.13.11/jars/actions.jar ...
	[SUCCESSFUL ] org.scala-sbt#actions;0.13.11!actions.jar (7368ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/main-settings/0.13.11/jars/main-settings.jar ...
	[SUCCESSFUL ] org.scala-sbt#main-settings;0.13.11!main-settings.jar (2967ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/interface/0.13.11/jars/interface.jar ...
	[SUCCESSFUL ] org.scala-sbt#interface;0.13.11!interface.jar (1933ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/io/0.13.11/jars/io.jar ...
	[SUCCESSFUL ] org.scala-sbt#io;0.13.11!io.jar (8875ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/ivy/0.13.11/jars/ivy.jar ...
	[SUCCESSFUL ] org.scala-sbt#ivy;0.13.11!ivy.jar (4027ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/logging/0.13.11/jars/logging.jar ...
	[SUCCESSFUL ] org.scala-sbt#logging;0.13.11!logging.jar (1999ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/logic/0.13.11/jars/logic.jar ...
	[SUCCESSFUL ] org.scala-sbt#logic;0.13.11!logic.jar (1924ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/process/0.13.11/jars/process.jar ...
	[SUCCESSFUL ] org.scala-sbt#process;0.13.11!process.jar (2013ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/run/0.13.11/jars/run.jar ...
	[SUCCESSFUL ] org.scala-sbt#run;0.13.11!run.jar (2014ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/command/0.13.11/jars/command.jar ...
	[SUCCESSFUL ] org.scala-sbt#command;0.13.11!command.jar (11828ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/launcher-interface/1.0.0-M1/launcher-interface-1.0.0-M1.jar ...
	[SUCCESSFUL ] org.scala-sbt#launcher-interface;1.0.0-M1!launcher-interface.jar (711ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/classpath/0.13.11/jars/classpath.jar ...
	[SUCCESSFUL ] org.scala-sbt#classpath;0.13.11!classpath.jar (2415ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/completion/0.13.11/jars/completion.jar ...
	[SUCCESSFUL ] org.scala-sbt#completion;0.13.11!completion.jar (7137ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/api/0.13.11/jars/api.jar ...
	[SUCCESSFUL ] org.scala-sbt#api;0.13.11!api.jar (4177ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/compiler-integration/0.13.11/jars/compiler-integration.jar ...
	[SUCCESSFUL ] org.scala-sbt#compiler-integration;0.13.11!compiler-integration.jar (4065ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/compiler-ivy-integration/0.13.11/jars/compiler-ivy-integration.jar ...
	[SUCCESSFUL ] org.scala-sbt#compiler-ivy-integration;0.13.11!compiler-ivy-integration.jar (2380ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/relation/0.13.11/jars/relation.jar ...
	[SUCCESSFUL ] org.scala-sbt#relation;0.13.11!relation.jar (6870ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/task-system/0.13.11/jars/task-system.jar ...
	[SUCCESSFUL ] org.scala-sbt#task-system;0.13.11!task-system.jar (9150ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/tasks/0.13.11/jars/tasks.jar ...
	[SUCCESSFUL ] org.scala-sbt#tasks;0.13.11!tasks.jar (9700ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/tracking/0.13.11/jars/tracking.jar ...
	[SUCCESSFUL ] org.scala-sbt#tracking;0.13.11!tracking.jar (2069ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/testing/0.13.11/jars/testing.jar ...
	[SUCCESSFUL ] org.scala-sbt#testing;0.13.11!testing.jar (1986ms)
downloading https://repo1.maven.org/maven2/org/scala-lang/scala-compiler/2.10.6/scala-compiler-2.10.6.jar ...
	[SUCCESSFUL ] org.scala-lang#scala-compiler;2.10.6!scala-compiler.jar (12985ms)
downloading https://repo1.maven.org/maven2/org/scala-lang/scala-reflect/2.10.6/scala-reflect-2.10.6.jar ...
	[SUCCESSFUL ] org.scala-lang#scala-reflect;2.10.6!scala-reflect.jar (3037ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/control/0.13.11/jars/control.jar ...
	[SUCCESSFUL ] org.scala-sbt#control;0.13.11!control.jar (2013ms)
downloading http://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/collections/0.13.11/jars/collections.jar ...
[... cut ...]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[error] (compile:compile) sbt.InvalidComponent: Could not find required component 'compiler-interface-src'
[error] Total time: 75 s, completed Mar 28, 2017 12:10:53 AM
[error] Not a valid command: v
[error] Not a valid project ID: v
[error] Expected ':' (if selecting a configuration)
[error] Not a valid key: v
[error] v
[error]  ^

kllbrd, 2017-03-27:
Since I can run them on my desktop I'm alright.
Thank you so much for your help!

mocapapa, 2017-12-19:
Same situation for me.
```
[sakurai@oak chisel-tutorial]$ sbt run
[info] Loading project definition from /home/sakurai/chisel/chisel-tutorial/project
[info] Set current project to chisel-tutorial (in build file:/home/sakurai/chisel/chisel-tutorial/)
[info] Compiling 56 Scala sources to /home/sakurai/chisel/chisel-tutorial/target/scala-2.11/classes...
sbt.InvalidComponent: Could not find required component 'compiler-interface-src'
        at sbt.ComponentManager.invalid(ComponentManager.scala:57)
        at sbt.ComponentManager.sbt$ComponentManager$$notFound$1(ComponentManager.scala:34)
        at sbt.ComponentManager.sbt$ComponentManager$$createAndCache$1(ComponentManager.scala:37)
        at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:27)
        at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:26)
        at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
        at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:48)
        at xsbt.boot.Locks$.apply0(Locks.scala:38)
        at xsbt.boot.Locks$.apply(Locks.scala:28)
        at sbt.ComponentManager.lock(ComponentManager.scala:50)
        at sbt.ComponentManager.lockGlobalCache(ComponentManager.scala:49)
        at sbt.ComponentManager.sbt$ComponentManager$$fromGlobal$1(ComponentManager.scala:25)
        at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
        at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
        at sbt.ComponentManager.sbt$ComponentManager$$getOrElse$1(ComponentManager.scala:32)
        at sbt.ComponentManager$$anonfun$files$1.apply(ComponentManager.scala:44)
        at sbt.ComponentManager$$anonfun$files$1.apply(ComponentManager.scala:44)
        at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
        at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:48)
        at xsbt.boot.Locks$.apply0(Locks.scala:38)
        at xsbt.boot.Locks$.apply(Locks.scala:28)
        at sbt.ComponentManager.lock(ComponentManager.scala:50)
        at sbt.ComponentManager.lockLocalCache(ComponentManager.scala:47)
        at sbt.ComponentManager.files(ComponentManager.scala:44)
        at sbt.compiler.ComponentCompiler$$anonfun$compileAndInstall$1.apply(ComponentCompiler.scala:61)
        at sbt.compiler.ComponentCompiler$$anonfun$compileAndInstall$1.apply(ComponentCompiler.scala:58)
        at sbt.IO$.withTemporaryDirectory(IO.scala:285)
        at sbt.compiler.ComponentCompiler.compileAndInstall(ComponentCompiler.scala:58)
        at sbt.compiler.ComponentCompiler$$anonfun$getLocallyCompiled$1.apply$mcV$sp(ComponentCompiler.scala:47)
        at sbt.IfMissing$Define.apply(ComponentManager.scala:77)
        at sbt.ComponentManager.sbt$ComponentManager$$createAndCache$1(ComponentManager.scala:39)
        at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:27)
        at sbt.ComponentManager$$anonfun$sbt$ComponentManager$$fromGlobal$1$1.apply(ComponentManager.scala:26)
        at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
        at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:98)
        at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:81)
        at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:102)
        at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:102)
        at xsbt.boot.Using$.withResource(Using.scala:11)
        at xsbt.boot.Using$.apply(Using.scala:10)
        at xsbt.boot.Locks$GlobalLock.withFileLock(Locks.scala:102)
        at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:62)
        at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:52)
        at xsbt.boot.Locks$.apply0(Locks.scala:38)
        at xsbt.boot.Locks$.apply(Locks.scala:28)
        at sbt.ComponentManager.lock(ComponentManager.scala:50)
        at sbt.ComponentManager.lockGlobalCache(ComponentManager.scala:49)
        at sbt.ComponentManager.sbt$ComponentManager$$fromGlobal$1(ComponentManager.scala:25)
        at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
        at sbt.ComponentManager$$anonfun$files$1$$anonfun$apply$2.apply(ComponentManager.scala:44)
        at sbt.ComponentManager.sbt$ComponentManager$$getOrElse$1(ComponentManager.scala:32)
        at sbt.ComponentManager$$anonfun$files$1.apply(ComponentManager.scala:44)
        at sbt.ComponentManager$$anonfun$files$1.apply(ComponentManager.scala:44)
        at sbt.ComponentManager$$anon$1.call(ComponentManager.scala:50)
        at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:98)
        at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:81)
        at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:102)
        at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:102)
        at xsbt.boot.Using$.withResource(Using.scala:11)
        at xsbt.boot.Using$.apply(Using.scala:10)
        at xsbt.boot.Locks$GlobalLock.withFileLock(Locks.scala:102)
        at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:62)
        at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:52)
[... 35 lines cut ...]
        at sbt.Defaults$.sbt$Defaults$$compileTaskImpl(Defaults.scala:736)
[... cut ...]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
[error] (compile:compile) sbt.InvalidComponent: Could not find required component 'compiler-interface-src'
[error] Total time: 0 s, completed 2017/12/20 8:38:54
[sakurai@oak chisel-tutorial]$ 
```
Command such as 'sbt -d -v clean run' gives me the same result.

ucbjrl, 2017-12-19:
Googling again indicates this may be a [Fedora problem](https://stackoverflow.com/questions/44330821/sbt-sbt-invalidcomponent-could-not-find-required-component-compiler-interfac).

mocapapa, 2017-12-20:
>Googling again indicates this may be a Fedora problem.

Thank you for your information! I am going to investigate the URL that you gave me.