ucb-bar/chisel-tutorial #83: Generated VCD

davidwende, 2017-04-27:
The problems/solutions are great.
Is there any way to generate vcd files for added usefulness?

There is some mention in the older documentation but this does not seem relevant for latest pull.

chick, 2017-04-27:
This is a little convoluted but here is the quick way.  We'll take advantage of the fact that the verilator backend creates a vcd output file by default, so we'll make it run instead of the default interpreter backend.  
Let's use the example of running the examples.Adder tutorial.  Run it as follows from the command line prompt.
```
TESTER_BACKENDS="verilator" ./run-examples.sh Adder
```
The vcd file will appear here
```
ls -l test_run_dir/examples.Launcher.*/*.vcd
```
I have added a more detailed explanation of another way to do this by using your own invocation of the Driver harness.  Check it out on the [Tutorial Wiki Faq](https://github.com/ucb-bar/chisel-tutorial/wiki/frequently-asked-questions)

chick, 2017-04-27:
@davidwende, this is a good question, I'd like recommend asking these kinds of questions on stack overflow using the [Chisel] tag. I think it provides a little better visibility for others who might be looking for this kind of answer

davidwende, 2017-04-30:
Hi Chick,

Tried the first solution and worked fine, except for one thing.
Verilator is not installed as a part of chisel-tutorial.
I installed the Linux distribution version instead (ARCH Linux) and this is
OK.
Thanks a lot for the help.
David