ucb-bar/chisel-tutorial #85: Documentation on tutorial wrong: missing GCD.scala

dleach, 2017-05-10:
Going through the tutorial documentation and the first example is missing from the directory location specified. "The Basics" page directs the user to src/main/scala/examples/ and open up GCD.scala but that doesn't exist. Fortunately the documentation includes the contents of the file but this should be clarified or cleaned up.

chick, 2017-05-11:
This needs to be fixed but I need to think about the best way.  GCD is a very nice example of chisel but it is also one of the tutorial problems.  I think if we put it in examples we should probably remove it from problems/solutions unless there is some overriding pedagogical reason for keeping it there

dleach, 2017-05-11:
I understand. I'm not a student and was instead looking for a good tutorial to try to get up to speed on chisel and was a bit disheartened when the first one in the set was wrong...

chick, 2017-07-06:
@dleach GCD was added back into examples in PR #86

closed 2017-07-06 by chick