ucb-bar/chisel-tutorial #87: Error in Using PeekPokeTester

MahdiNazemi, 2017-05-31:
When I try to run a script like run-examples.sh, I get the following errors for all modules:

[error] $dir/chisel-tutorial/src/test/scala/problems/Accumulator.scala:6: inferred type arguments [problems.Accumulator] do not conform to type PeekPokeTester's type parameter bounds [+T <: Chisel.Module]
[error] Error occurred in an application involving default arguments.
[error] class AccumulatorTests(c: Accumulator) extends PeekPokeTester(c) {
[error]                                                ^
[error] $dir/chisel-tutorial/src/test/scala/problems/Accumulator.scala:6: type mismatch;
[error]  found   : problems.Accumulator
[error]  required: T
[error] Error occurred in an application involving default arguments.
[error] class AccumulatorTests(c: Accumulator) extends PeekPokeTester(c) {
[error]                                                               ^

Do I need to change anything to make the tutorials work again?

chick, 2017-05-31:
Sorry to hear this.  I suspect this is some sort of mismatched versioning problem. I was able to able run. After the following steps on a machine like this
Linux a6 4.4.0-72-generic #93-Ubuntu SMP Fri Mar 31 14:07:41 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
After the following steps
```
rm -rf ~/.ivy2/cache/edu.berkeley.cs//*
rm -rf ~/.ivy2/local/edu.berkeley.cs//*
mkdir workspace
cd workspace
git clone https://github.com/ucb-bar/chisel-tutorial.git
./run-examples.sh
```
So questions:
Are your running the release branch or the master branch (release is the defa?ult)
Have you tried clearing the dependencies, as with the rm commands shown?

MahdiNazemi, 2017-06-01:
Thank you for your prompt reply. I followed the steps you mentioned and the problem is gone. However, running the following commands was not enough.

`rm -rf ~/.ivy2/cache/edu.berkeley.cs/*
rm -rf ~/.ivy2/local/edu.berkeley.cs/*`

I had to remove the local repo and clone again to solve the issue.

Do you know what could have been different between my local copy and the release branch that caused this issue? I get the same errors in my own chisel projects and would like to know how to solve them.

chick, 2017-06-05:
Sorry for the delay, I've been asking around here.  I think it's likely related to change in the Module hierarchy #469.  Do you have sharable code that exhibits the problem

MahdiNazemi, 2017-06-09:
Sorry for my late reply. Attached, please find the implementation of floating-point multiplication.

The testing mechanism is similar to the TutorialRunner in this package, so the commands will look like this:
`sbt`
`test:run-main easi.Launcher FPMul`

[FPMul.zip](https://github.com/ucb-bar/chisel-tutorial/files/1065367/FPMul.zip)

MahdiNazemi, 2017-06-09:
One of the differences I noticed is that the line
`import Chisel.iotesters.{ChiselFlatSpec, Driver, PeekPokeTester}`
in previous implementations is changed to
`import chisel3.iotesters.{ChiselFlatSpec, Driver, PeekPokeTester}`.

But even after applying the same modification, I get 
`object iotesters is not a member of package chisel3`.

MahdiNazemi, 2017-06-09:
After making the change mentioned in the previous comment, I did
`rm -rf ~/.ivy2/cache/edu.berkeley.cs/*`
` rm -rf ~/.ivy2/local/edu.berkeley.cs/*`
`sbt clean`
and the issue is gone.

So I guess the issue was related to both mismatched versions and changing Chisel to chisel3 in import statements.

closed 2017-06-09 by MahdiNazemi

chick, 2017-06-09:
@MahdiNazemi 
Glad to hear it, I just compiled it all here and I think you got it.
- Definitely try to use import chisel3._ instead of capital chisel.
- I found it difficult to have sbt sort things out given the directory hierarchy.  I'd recommend using the layout found in the chisel-template repo.  I think it's a bit convoluted for very simple tests but the tools have expectations and conventions and they work better the more those are kept too
- sbt, does a lot of caching, after I had move things around, it seemed to take an exit from the sbt shell and a clean command before things started working