ucb-bar/chisel-tutorial #95: when is a combinatorial construct

schoeberl, 2017-08-28:
I think the following text is missleading:

"The when statement tells Chisel to perform the operation on a positive clock edge if the condition is true, treating the left hand assignments as synchronous. This is similar to how Verilog uses always @ (posedge clk) to specify synchronous logic."

when is a combinational construct, basically a multiplexer. Only registers "perform an operation on a positive clock edge".

schoeberl, 2017-08-28:
I am surprised that I can edit the wiki pages. Can anyone do that or do I have write access?

Shall I go ahead and rewrite that paragraph?

aswaterman, 2017-08-28:
I'm not sure about the permissions issue, but I think a correct explanation of ```when``` depends on what's on the left-hand side of the assignment.  If it's a register, the analogy to ```always @(posedge clk)``` is accurate.  If it's a wire, it's combinational, like you said.

schoeberl, 2017-08-28:
When is a condition, like if in VHDL, that results in a multiplexer in hardware. After when, in the true block, there can be more than one assignment. Be those to a wire or a register. Actually this is an interesting difference between VHDL (and probably Verilog as well) that one when (if) can be used for both combinational and synchron destinantions. Maybe not so super easy to describe this consice, but exact in a tutorial.