ucb-bar/chisel-tutorial #105: Cann't generate verilog

zhangyl4991, 2017-10-19:
Hi, when i follow the Wiki try to generate verilog: TESTER_BACKENDS=verilator ./run-examples.sh GCD. It only generate GCD.anno, GCD.fir and GCD.lo.fir, and i get the following information:
[process_args] java_version = '1.7'
# Executing command line:
java
-Xms1024m
-Xmx1024m
-XX:ReservedCodeCacheSize=128m
-XX:MaxPermSize=256m
-jar
/usr/share/sbt/bin/sbt-launch.jar
"test:run-main examples.Launcher GCD"

[info] Loading project definition from /home/zhangyl/Project/chisel/chisel-tutorial/project
[info] Set current project to chisel-tutorial (in build file:/home/zhangyl/Project/chisel/chisel-tutorial/)
[info] Running examples.Launcher GCD
Starting tutorial GCD
[info] [0.004] Elaborating design...
[info] [0.282] Done elaborating.
Total FIRRTL Compile Time: 2140.0 ms
Total FIRRTL Compile Time: 68.9 ms
End of dependency graph
Circuit state created
[info] [0.001] SEED 1508412833942
test GCD Success: 3 tests passed in 29 cycles taking 0.057114 seconds
[info] [0.042] RAN 24 CYCLES PASSED
Tutorials passing: 1
[success] Total time: 6 s, completed Oct 19, 2017 7:33:57 PM

My ubuntu's version is 14.04, and i have installed verilator.

ucbjrl, 2017-10-19:
Unfortunately, the wiki documentation is slightly out-of-date. We'll update it.

The way to generate Verilog with the current tutorial release is:
```bash
./run-examples.sh GCD --backend-name verilator
```

ucbjrl, 2017-10-19:
Documentation updated.

closed 2017-10-19 by ucbjrl