ucb-bar/chisel-tutorial #111: Facing issue while generating verilog

Sairam-Ganti, 2017-12-17:
Following the steps [here](https://github.com/ucb-bar/chisel-tutorial/wiki/the-basics) 
At the "Generating Verilog" step when I run `./run-examples.sh GCD --backend-name verilator` command, I get an error like so: 

`D --backend-name verilator
[process_args] java_version = '1.8.0_151'
# Executing command line:
java
-Xms1024m
-Xmx1024m
-XX:ReservedCodeCacheSize=128m
-jar
/usr/share/sbt-launcher-packaging/bin/sbt-launch.jar
"test:run-main examples.Launcher GCD --backend-name verilator"

[info] Loading project definition from /home/sairamganti/Documents/devenv/chisel-tutorial/project
[info] Set current project to chisel-tutorial (in build file:/home/sairamganti/Documents/devenv/chisel-tutorial/)
[info] Running examples.Launcher GCD --backend-name verilator
Starting tutorial GCD
[info] [0.001] Elaborating design...
[info] [0.076] Done elaborating.
Total FIRRTL Compile Time: 878.7 ms
verilator --cc /home/sairamganti/Documents/devenv/chisel-tutorial/test_run_dir/examples/GCD/GCD.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O1 --top-module GCD +define+TOP_TYPE=VGCD +define+PRINTF_COND=!GCD.reset +define+STOP_COND=!GCD.reset -CFLAGS -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h -Mdir /home/sairamganti/Documents/devenv/chisel-tutorial/test_run_dir/examples/GCD --exe /home/sairamganti/Documents/devenv/chisel-tutorial/test_run_dir/examples/GCD/GCD-harness.cpp
make: Entering directory '/home/sairamganti/Documents/devenv/chisel-tutorial/test_run_dir/examples/GCD'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h   -c -o GCD-harness.o /home/sairamganti/Documents/devenv/chisel-tutorial/test_run_dir/examples/GCD/GCD-harness.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h   -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h   -c -o verilated_vcd_c.o /usr/local/share/verilator/include/verilated_vcd_c.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VGCD.cpp > VGCD__ALLcls.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VGCD__Trace.cpp VGCD__Syms.cpp VGCD__Trace__Slow.cpp > VGCD__ALLsup.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h   -c -o VGCD__ALLcls.o VGCD__ALLcls.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VGCD -DVL_USER_FINISH -include VGCD.h   -c -o VGCD__ALLsup.o VGCD__ALLsup.cpp
      Archiving VGCD__ALL.a ...
ar r VGCD__ALL.a VGCD__ALLcls.o VGCD__ALLsup.o
ranlib VGCD__ALL.a
g++    GCD-harness.o verilated.o verilated_vcd_c.o VGCD__ALL.a    -o VGCD -lm -lstdc++  2>&1 | c++filt
verilated.o: In function `VL_FINISH_MT(char const*, int, char const*)':
verilated.cpp:(.text+0x7): undefined reference to `vl_finish(char const*, int, char const*)'
collect2: error: ld returned 1 exit status
make: Leaving directory '/home/sairamganti/Documents/devenv/chisel-tutorial/test_run_dir/examples/GCD'
java.lang.IllegalArgumentException: requirement failed: test_run_dir/examples/GCD/VGCD doesn't exist
	at scala.Predef$.require(Predef.scala:224)
	at chisel3.iotesters.TesterProcess$.apply(PeekPokeTesterUtils.scala:153)
	at chisel3.iotesters.SimApiInterface.<init>(SimApiInterface.scala:36)
	at chisel3.iotesters.VerilatorBackend.<init>(VerilatorBackend.scala:281)
	at chisel3.iotesters.setupVerilatorBackend$.apply(VerilatorBackend.scala:270)
	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply$mcZ$sp(Driver.scala:53)
	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:38)
	at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:38)
	at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:129)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at logger.Logger$.makeScope(Logger.scala:127)
	at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala:38)
	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:38)
	at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:38)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.iotesters.Driver$.execute(Driver.scala:37)
	at examples.Launcher$$anonfun$13.apply(Launcher.scala:70)
	at examples.Launcher$$anonfun$13.apply(Launcher.scala:69)
	at utils.TutorialRunner$$anonfun$apply$2.apply(TutorialRunner.scala:43)
	at utils.TutorialRunner$$anonfun$apply$2.apply(TutorialRunner.scala:36)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at utils.TutorialRunner$.apply(TutorialRunner.scala:36)
	at examples.Launcher$.main(Launcher.scala:111)
	at examples.Launcher.main(Launcher.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sbt.Run.invokeMain(Run.scala:67)
	at sbt.Run.run0(Run.scala:61)
	at sbt.Run.sbt$Run$$execute$1(Run.scala:51)
	at sbt.Run$$anonfun$run$1.apply$mcV$sp(Run.scala:55)
	at sbt.Run$$anonfun$run$1.apply(Run.scala:55)
	at sbt.Run$$anonfun$run$1.apply(Run.scala:55)
	at sbt.Logger$$anon$4.apply(Logger.scala:84)
	at sbt.TrapExit$App.run(TrapExit.scala:248)
	at java.lang.Thread.run(Thread.java:748)
================================================================================
Errors: 1: in the following tutorials
Tutorial GCD: exception requirement failed: test_run_dir/examples/GCD/VGCD doesn't exist
================================================================================

`

Any help?

ucbjrl, 2017-12-18:
@Sairam-Ganti this looks like a chisel3 version issue. Is it possible you have an old version of the chisel-tutorial repository? Please type the following command in the chisel-tutorial directory:
```bash
$ git log -1
```
This should give the last commit for this clone of the chisel-tutorial branch. A:
```bash
$ git checkout release
$ git pull
```
should get you the latest version, after which an:
```bash
$ sbt clean "test:run-main examples.Launcher GCD --backend-name verilator"
```
should work. If this still fails, could you issue the following command:
```bash
$ sbt clean "test:run-main examples.Launcher GCD --backend-name verilator" >& sbttest.out
```
and attach the generated `sbttest.out` file.

Sairam-Ganti, 2017-12-19:
@ucbjrl 
`git log -1` gave me the following output (Note: I'm already at the latest version)

`commit 1f8d68e1f40c03ff9fd8e90739bc9cef9234fdfa
Author: Jim Lawson <ucbjrl@berkeley.edu>
Date:   Wed Nov 29 13:10:51 2017 -0800

    Bump release versions (#109)
    
    * InvalidateAPI - Initialize all router outputs.
    
    * Add "last connect" semantics comment.
    
    * Fix comment in Adder (problem and solution).
    
    * Enable cross-compile. (#108)
`
And, it failed again, attaching the sbttest.out file here. 
[sbttest.zip](https://github.com/ucb-bar/chisel-tutorial/files/1571950/sbttest.zip)

ucbjrl, 2017-12-19:
Thanks @Sairam-Ganti. This looks like a chisel3/verilator version mismatch. Which version of verilator are you using?

ucbjrl, 2017-12-19:
I've verified that this is an issue with newer versions of verilator. Until we update chisel3 to work with recent verilator releases, I suggest you use: Verilator 3.904 2017-05-30

Sairam-Ganti, 2017-12-20:
@ucbjrl  Thanks, will switch to the specified version and update with the result

Sairam-Ganti, 2017-12-26:
Hi, so I switched to the specified version of verilator and now I get the following error: 
`[info] Loading project definition from /home/sairamganti/Documents/devenv/chisel-tutorial/project
[info] Set current project to chisel-tutorial (in build file:/home/sairamganti/Documents/devenv/chisel-tutorial/)
[info] Updating {file:/home/sairamganti/Documents/devenv/chisel-tutorial/}chisel-tutorial...
[info] Resolving jline#jline;2.14.3 ...
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: configuration not found in edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: 'master'. It was required from edu.berkeley.cs#chisel-iotesters_2.11;1.1.1 compile
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:chisel3_2.11:[3.0,3.1)
[warn] 		  +- edu.berkeley.cs:chisel-iotesters_2.11:1.1.+ (/home/sairamganti/Documents/devenv/chisel-tutorial/build.sbt#L46-48)
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.1-SNAPSHOT
sbt.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;[3.0,3.1): configuration not found in edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: 'master'. It was required from edu.berkeley.cs#chisel-iotesters_2.11;1.1.1 compile
	at sbt.IvyActions$.sbt$IvyActions$$resolve(IvyActions.scala:313)
	at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:191)
	at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:168)
	at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:156)
	at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:156)
	at sbt.IvySbt$$anonfun$withIvy$1.apply(Ivy.scala:133)
	at sbt.IvySbt.sbt$IvySbt$$action$1(Ivy.scala:57)
	at sbt.IvySbt$$anon$4.call(Ivy.scala:65)
	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
	at xsbt.boot.Using$.withResource(Using.scala:10)
	at xsbt.boot.Using$.apply(Using.scala:9)
	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
	at xsbt.boot.Locks$.apply0(Locks.scala:31)
	at xsbt.boot.Locks$.apply(Locks.scala:28)
	at sbt.IvySbt.withDefaultLogger(Ivy.scala:65)
	at sbt.IvySbt.withIvy(Ivy.scala:128)
	at sbt.IvySbt.withIvy(Ivy.scala:125)
	at sbt.IvySbt$Module.withModule(Ivy.scala:156)
	at sbt.IvyActions$.updateEither(IvyActions.scala:168)
	at sbt.Classpaths$$anonfun$sbt$Classpaths$$work$1$1.apply(Defaults.scala:1541)
	at sbt.Classpaths$$anonfun$sbt$Classpaths$$work$1$1.apply(Defaults.scala:1537)
	at sbt.Classpaths$$anonfun$doWork$1$1$$anonfun$121.apply(Defaults.scala:1572)
	at sbt.Classpaths$$anonfun$doWork$1$1$$anonfun$121.apply(Defaults.scala:1570)
	at sbt.Tracked$$anonfun$lastOutput$1.apply(Tracked.scala:37)
	at sbt.Classpaths$$anonfun$doWork$1$1.apply(Defaults.scala:1575)
	at sbt.Classpaths$$anonfun$doWork$1$1.apply(Defaults.scala:1569)
	at sbt.Tracked$$anonfun$inputChanged$1.apply(Tracked.scala:60)
	at sbt.Classpaths$.cachedUpdate(Defaults.scala:1592)
	at sbt.Classpaths$$anonfun$updateTask$1.apply(Defaults.scala:1519)
	at sbt.Classpaths$$anonfun$updateTask$1.apply(Defaults.scala:1471)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
[error] (*:update) sbt.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;[3.0,3.1): configuration not found in edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: 'master'. It was required from edu.berkeley.cs#chisel-iotesters_2.11;1.1.1 compile
`
It seems like it cant find the "3.1-SNAPSHOT" from chisel-iotesters.

ucbjrl, 2017-12-26:
Based on the unresolved dependencies, it looks like you've trying to use the `master` branch version of the tutorials. You should be using the `release` (the default) branch, which uses the published versions of the chisel dependencies.
```bash
% cd <...>/chisel-tutorial
% git checkout release
% git pull
% sbt test
```
You may use the `master` branch, but if you do so, you need to clone the chisel dependencies ([firrtl](https://github.com/freechipsproject/firrtl), [firrtl-interpreter](https://github.com/freechipsproject/firrtl-interpreter), [chisel3](https://github.com/freechipsproject/chisel3), and [chisel-testers](https://github.com/freechipsproject/chisel-testers)), and `publishLocal` each of them in that order.
The `master` branch is targeted towards those developers actively developing chisel tutorial code. The `release` branch is for those using the chisel tutorial to learn about chisel.