ucb-bar/chisel-tutorial #112: problem in build.sbt

poweihuang17, 2018-01-13:
Hi, when I issue " sbt run", it generated this kind of error.
My sbt version is 1.1.0.
And my setting in build.sbt  is like this:

name := "chisel-tutorial"

scalaVersion := "2.12.4"

crossScalaVersions := Seq("2.11.11", "2.12.3")
-------
/Users/Powei/Dropbox/riscv/chisel/chisel-tutorial/build.sbt:7: error: scrutinee is incompatible with pattern type;
 found   : Int
 required: Long
      case Some((2, scalaMajor: Int)) if scalaMajor < 12 => Seq()
                                ^
/Users/Powei/Dropbox/riscv/chisel/chisel-tutorial/build.sbt:19: error: scrutinee is incompatible with pattern type;
 found   : Int
 required: Long
      case Some((2, scalaMajor: Int)) if scalaMajor < 12 =>

poweihuang17, 2018-01-13:
I fix it by changing the type from Int to Long.

ucbjrl, 2018-01-13:
The release branches of chisel-template and chisel-tutorial use sbt 0.13. When we switch to sbt 1.0 and Scala 2.12 (which will probably happen within a month) for all the chisel-related repos, we'll update these as well.

ucbjrl, 2018-04-18:
Closed by #123

closed 2018-04-18 by ucbjrl