ucb-bar/chisel-tutorial #124: sbt run problem

kaito0422, 2018-04-20:
Hello,
I did as the tutorial does. And I got a problem. When I sbt run, I got many errors

gjc@302f:~/chisel-tutorial$ sbt run
[info] Loading project definition from /home/gjc/chisel-tutorial/project
[info] Updating ProjectRef(uri("file:/home/gjc/chisel-tutorial/project/"), "chisel-tutorial-build")...
[info] Done updating.
[info] Loading settings from build.sbt ...
[info] Set current project to chisel-tutorial (in build file:/home/gjc/chisel-tutorial/)
[info] Updating ...
[warn] 	module not found: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/gjc/.ivy2/local/edu.berkeley.cs/chisel3_2.11/3.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.11/3.2-SNAPSHOT/chisel3_2.11-3.2-SNAPSHOT.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/gjc/.sbt/preloaded/edu.berkeley.cs/chisel3_2.11/3.2-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/gjc/.sbt/preloaded/edu/berkeley/cs/chisel3_2.11/3.2-SNAPSHOT/chisel3_2.11-3.2-SNAPSHOT.pom
[warn] ==== sonatype-snapshots: tried
[warn]   https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.11/3.2-SNAPSHOT/chisel3_2.11-3.2-SNAPSHOT.pom
[warn] ==== sonatype-releases: tried
[warn]   https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel3_2.11/3.2-SNAPSHOT/chisel3_2.11-3.2-SNAPSHOT.pom
[warn] 	module not found: edu.berkeley.cs#chisel-iotesters_2.11;1.3-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/gjc/.ivy2/local/edu.berkeley.cs/chisel-iotesters_2.11/1.3-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/chisel-iotesters_2.11/1.3-SNAPSHOT/chisel-iotesters_2.11-1.3-SNAPSHOT.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/gjc/.sbt/preloaded/edu.berkeley.cs/chisel-iotesters_2.11/1.3-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/gjc/.sbt/preloaded/edu/berkeley/cs/chisel-iotesters_2.11/1.3-SNAPSHOT/chisel-iotesters_2.11-1.3-SNAPSHOT.pom
[warn] ==== sonatype-snapshots: tried
[warn]   https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel-iotesters_2.11/1.3-SNAPSHOT/chisel-iotesters_2.11-1.3-SNAPSHOT.pom
[warn] ==== sonatype-releases: tried
[warn]   https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel-iotesters_2.11/1.3-SNAPSHOT/chisel-iotesters_2.11-1.3-SNAPSHOT.pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT: not found
[warn] 	:: edu.berkeley.cs#chisel-iotesters_2.11;1.3-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:chisel3_2.11:3.2-SNAPSHOT (/home/gjc/chisel-tutorial/build.sbt#L46-48)
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.2-SNAPSHOT
[warn] 		edu.berkeley.cs:chisel-iotesters_2.11:1.3-SNAPSHOT (/home/gjc/chisel-tutorial/build.sbt#L46-48)
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.2-SNAPSHOT
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT: not found
[error] unresolved dependency: edu.berkeley.cs#chisel-iotesters_2.11;1.3-SNAPSHOT: not found
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:205)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:243)
[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
[error] 	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
[error] 	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[error] 	at xsbt.boot.Using$.withResource(Using.scala:10)
[error] 	at xsbt.boot.Using$.apply(Using.scala:9)
[error] 	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
[error] 	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
[error] 	at xsbt.boot.Locks$.apply0(Locks.scala:31)
[error] 	at xsbt.boot.Locks$.apply(Locks.scala:28)
[error] 	at sbt.internal.librarymanagement.IvySbt.withDefaultLogger(Ivy.scala:77)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:199)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:196)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.withModule(Ivy.scala:242)
[error] 	at sbt.internal.librarymanagement.IvyActions$.updateEither(IvyActions.scala:190)
[error] 	at sbt.librarymanagement.ivy.IvyDependencyResolution.update(IvyDependencyResolution.scala:20)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:56)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:46)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:99)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:68)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$19(LibraryManagement.scala:112)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:224)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:112)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:95)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:149)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:126)
[error] 	at sbt.Classpaths$.$anonfun$updateTask$5(Defaults.scala:2383)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:66)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:271)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (update) sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;3.2-SNAPSHOT: not found
[error] unresolved dependency: edu.berkeley.cs#chisel-iotesters_2.11;1.3-SNAPSHOT: not found
[error] Total time: 80 s, completed Apr 20, 2018 10:07:44 PM
gjc@302f:~/chisel-tutorial$ 

anyone can help me, please
Thank you.

edwardcwang, 2018-04-20:
Are you on the master branch?

ucbjrl, 2018-04-20:
The master branch `README.md` was incorrectly merged with the release branch version. I've updated it to reflect the correct instructions for the release version.

Please checkout the release version of the chisel-tutorial:
```bash
git checkout release
```

kaito0422, 2018-04-21:
@ucbjrl 
Thank you, Jim. I did git checkout release and I got [success]. 
But I don't know what is master branch. 
It's the first time for me to use this tool. I think I need some preliminary knowledge. Can you give me some advice, please?

ucbjrl, 2018-04-23:
Git (and GitHub) support multiple _branches_ (versions) of software. The `master` branch is where most development work is consolidated in our repositories. The `master` branch is typically in a continual state of flux, and may require significant effort to work with other repositories (which are also typically in a state of flux). If you're working in the `master` branch, we assume you're _developing_ the chisel-tutorial (as opposed to simply _using_ it), and you've checked out and locally published development versions of its dependencies. The `release` branch is much more stable, and has been tested with stable versions of its dependencies. **Note**: this is purely by convention. Other organizations have different policies.

I'm going to close this issue now (thanks for bringing it to our attention) as I believe it was resolved by 4d777ab, but feel free to raise another issue if you feel we need to provide more documentation on either _using_ or _developing_ the chisel tutorial.

If you have questions about `git` or `sbt`, a web search should point you at documentation for those tools.

closed 2018-04-23 by ucbjrl

yonghaoXuGit, 2018-11-09:
I encountered the same problem,But why can't find release branch?

ucbjrl, 2018-11-09:
The release branch should be the default branch when you clone the
repository. If you type:

git branch -v

in the root of your copy of the chisel-tutorial repository, it should tell
you which branches are available. The branch you're currently on should be
marked with an asterisk.
If you would like more assistance, we need to know what you've tried and
what the results were.

Look at the initial report of this issue
<https://github.com/ucb-bar/chisel-tutorial/issues/124> and see the text
(not a screen shot) that was supplied. We need something like that in order
to suggest how to resolve this.

yonghaoXuGit, 2018-11-21:
Ok,Thanks,I find correct bran using git branch -v.

来自 Outlook<http://aka.ms/weboutlook>