ucb-bar/chisel-tutorial #128: sbt run error

Arna-Maity, 2018-05-16:
I get a similar problem as #124 even while i am on the release branch.
Could anyone please help me ?
I am using Ubuntu 18.04 and sbt 1.1.1
Below is the output -

[info] Loading project definition from /home/arna/chisel-tutorial/project
[info] Loading settings from build.sbt ...
[info] Set current project to chisel-tutorial (in build file:/home/arna/chisel-tutorial/)
[info] Updating ...
[warn] 	module not found: edu.berkeley.cs#chisel3_2.11;3.1.+
[warn] ==== local: tried
[warn]   /home/arna/.ivy2/local/edu.berkeley.cs/chisel3_2.11/[revision]/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.11/[revision]/chisel3_2.11-[revision].pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/arna/.sbt/preloaded/edu.berkeley.cs/chisel3_2.11/[revision]/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/arna/.sbt/preloaded/edu/berkeley/cs/chisel3_2.11/[revision]/chisel3_2.11-[revision].pom
[warn] ==== sonatype-snapshots: tried
[warn]   https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.11/[revision]/chisel3_2.11-[revision].pom
[warn] ==== sonatype-releases: tried
[warn]   https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel3_2.11/[revision]/chisel3_2.11-[revision].pom
[warn] 	module not found: edu.berkeley.cs#chisel-iotesters_2.11;1.2.+
[warn] ==== local: tried
[warn]   /home/arna/.ivy2/local/edu.berkeley.cs/chisel-iotesters_2.11/[revision]/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/chisel-iotesters_2.11/[revision]/chisel-iotesters_2.11-[revision].pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/arna/.sbt/preloaded/edu.berkeley.cs/chisel-iotesters_2.11/[revision]/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/arna/.sbt/preloaded/edu/berkeley/cs/chisel-iotesters_2.11/[revision]/chisel-iotesters_2.11-[revision].pom
[warn] ==== sonatype-snapshots: tried
[warn]   https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel-iotesters_2.11/[revision]/chisel-iotesters_2.11-[revision].pom
[warn] ==== sonatype-releases: tried
[warn]   https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel-iotesters_2.11/[revision]/chisel-iotesters_2.11-[revision].pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: org.scala-lang#scala-library;2.11.12: Resolution failed several times for dependency: org.scala-lang#scala-library;2.11.12 {scala-tool=[default, optional(default)], compile=[default(compile)]}:: 
[warn] 	public: unable to get resource for org/scala-lang#scala-library;2.11.12: res=https://repo1.maven.org/maven2/org/scala-lang/scala-library/2.11.12/scala-library-2.11.12.pom: java.net.ConnectException: Failed to connect to repo1.maven.org/64:ff9b:0:0:0:0:9765:28d1:443
[warn] 	sonatype-snapshots: unable to get resource for org/scala-lang#scala-library;2.11.12: res=https://oss.sonatype.org/content/repositories/snapshots/org/scala-lang/scala-library/2.11.12/scala-library-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[warn] 	sonatype-releases: unable to get resource for org/scala-lang#scala-library;2.11.12: res=https://oss.sonatype.org/content/repositories/releases/org/scala-lang/scala-library/2.11.12/scala-library-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[warn] 	:: edu.berkeley.cs#chisel3_2.11;3.1.+: not found
[warn] 	:: edu.berkeley.cs#chisel-iotesters_2.11;1.2.+: not found
[warn] 	:: org.scala-lang#scala-compiler;2.11.12: Resolution failed several times for dependency: org.scala-lang#scala-compiler;2.11.12 {scala-tool=[default, optional(default)]}:: 
[warn] 	public: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://repo1.maven.org/maven2/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to repo1.maven.org/64:ff9b:0:0:0:0:9765:28d1:443
[warn] 	sonatype-snapshots: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://oss.sonatype.org/content/repositories/snapshots/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[warn] 	sonatype-releases: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://oss.sonatype.org/content/repositories/releases/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		org.scala-lang:scala-library:2.11.12 ((sbt.Classpaths.jvmBaseSettings) Defaults.scala#L2055)
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.1.0
[warn] 		edu.berkeley.cs:chisel3_2.11:3.1.+ (/home/arna/chisel-tutorial/build.sbt#L46-48)
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.1.0
[warn] 		edu.berkeley.cs:chisel-iotesters_2.11:1.2.+ (/home/arna/chisel-tutorial/build.sbt#L46-48)
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.1.0
[warn] 		org.scala-lang:scala-compiler:2.11.12
[warn] 		  +- edu.berkeley.cs:chisel-tutorial_2.11:3.1.0
[error] sbt.librarymanagement.ResolveException: unresolved dependency: org.scala-lang#scala-library;2.11.12: Resolution failed several times for dependency: org.scala-lang#scala-library;2.11.12 {scala-tool=[default, optional(default)], compile=[default(compile)]}:: 
[error] 	public: unable to get resource for org/scala-lang#scala-library;2.11.12: res=https://repo1.maven.org/maven2/org/scala-lang/scala-library/2.11.12/scala-library-2.11.12.pom: java.net.ConnectException: Failed to connect to repo1.maven.org/64:ff9b:0:0:0:0:9765:28d1:443
[error] 	sonatype-snapshots: unable to get resource for org/scala-lang#scala-library;2.11.12: res=https://oss.sonatype.org/content/repositories/snapshots/org/scala-lang/scala-library/2.11.12/scala-library-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[error] 	sonatype-releases: unable to get resource for org/scala-lang#scala-library;2.11.12: res=https://oss.sonatype.org/content/repositories/releases/org/scala-lang/scala-library/2.11.12/scala-library-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[error] 
[error] unresolved dependency: edu.berkeley.cs#chisel3_2.11;3.1.+: not found
[error] unresolved dependency: edu.berkeley.cs#chisel-iotesters_2.11;1.2.+: not found
[error] unresolved dependency: org.scala-lang#scala-compiler;2.11.12: Resolution failed several times for dependency: org.scala-lang#scala-compiler;2.11.12 {scala-tool=[default, optional(default)]}:: 
[error] 	public: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://repo1.maven.org/maven2/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to repo1.maven.org/64:ff9b:0:0:0:0:9765:28d1:443
[error] 	sonatype-snapshots: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://oss.sonatype.org/content/repositories/snapshots/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[error] 	sonatype-releases: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://oss.sonatype.org/content/repositories/releases/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[... 35 lines cut ...]
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:66)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:271)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[... cut ...]
[error] 	sonatype-snapshots: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://oss.sonatype.org/content/repositories/snapshots/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[error] 	sonatype-releases: unable to get resource for org/scala-lang#scala-compiler;2.11.12: res=https://oss.sonatype.org/content/repositories/releases/org/scala-lang/scala-compiler/2.11.12/scala-compiler-2.11.12.pom: java.net.ConnectException: Failed to connect to oss.sonatype.org/64:ff9b:0:0:0:0:22c7:6fed:443
[error] Total time: 26 s, completed 16 May, 2018 10:10:58 AM

ucbjrl, 2018-05-16:
This looks like network connectivity/DNS issues. You don't seem to be able to access the Maven (https://repo1.maven.org) or Sonatype (https://oss.sonatype.org) repositories.

Arna-Maity, 2018-05-16:
But i tried executing the same commands while on different networks and all yielded the same result.

Arna-Maity, 2018-05-16:
I also executed the same commands on another computer running Ubuntu 16.04 on the same network and it executed successfully.

ucbjrl, 2018-05-16:
You need to find out what the difference is between the machines on which the build works and those on which it fails. Your system or network administrators may be helpful.

Arna-Maity, 2018-09-12:
The issue was successfully resolved after few software  updates. Thanks @ucbjrl

closed 2018-09-12 by Arna-Maity