ucb-bar/chisel-tutorial #129: java.lang.Runtimeexception

jjtoms4s, 2018-05-28:
[image]
i follow the quick_start and then i run into this error when run this command ( test:run-main problems.Launcher Mux4) does anyone have idea

jjtoms4s, 2018-05-28:
in addtions the MUX2 work but the other task run into the same error

ucbjrl, 2018-05-29:
The README.md was out of date for the updated `sbt`. We've updated it to specify the correct invocation for running tests.

Additionally, the initial `problems` tests are expected to generate errors, and we force an early exit with a non-zero exit code to indicate failure (the tests will fail until you add the correct code to make them pass). The updated `sbt` was throwing a security exception in response to this attempt to exit early.

Fixed by #125 and #131.

closed 2018-05-29 by ucbjrl