ucb-bar/chisel-tutorial #133: error with sbt run

eahmedsalman, 2018-07-31:
at sbt.internal.inc.IncrementalCompilerImpl.handleCompilationError(IncrementalCompilerImpl.scala:158)
        at sbt.internal.inc.IncrementalCompilerImpl.compileIncrementally(IncrementalCompilerImpl.scala:237)
        at sbt.internal.inc.IncrementalCompilerImpl.compile(IncrementalCompilerImpl.scala:68)
        at sbt.Defaults$.compileIncrementalTaskImpl(Defaults.scala:1429)
        at sbt.Defaults$.$anonfun$compileIncrementalTask$1(Defaults.scala:1403)
        at scala.Function1.$anonfun$compose$1(Function1.scala:44)
        at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
        at sbt.std.Transform$$anon$4.work(System.scala:66)
        at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
        at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
        at sbt.Execute.work(Execute.scala:271)
        at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
        at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
        at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
        at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
[info] Attempting to fetch org.scala-sbt:compiler-bridge_2.11:1.1.1.
[error] ## Exception when compiling 56 sources to C:\Users\salman\chisel-tutorial\target\scala-2.11\classes
[error] null
[error] java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
[error] java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
[error] java.base/java.lang.reflect.Method.invoke(Unknown Source)
[error] sbt.internal.inc.RawCompiler.getReporter$1(RawCompiler.scala:50)
[error] sbt.internal.inc.RawCompiler.apply(RawCompiler.scala:71)
[error] sbt.internal.inc.AnalyzingCompiler$.$anonfun$compileSources$6(AnalyzingCompiler.scala:350)
[error] sbt.internal.inc.AnalyzingCompiler$.handleCompilationError$1(AnalyzingCompiler.scala:327)
[error] sbt.internal.inc.AnalyzingCompiler$.$anonfun$compileSources$4(AnalyzingCompiler.scala:346)
[error] sbt.internal.inc.AnalyzingCompiler$.$anonfun$compileSources$4$adapted(AnalyzingCompiler.scala:341)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:376)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:383)
[error] sbt.internal.inc.AnalyzingCompiler$.$anonfun$compileSources$2(AnalyzingCompiler.scala:341)
[error] sbt.internal.inc.AnalyzingCompiler$.$anonfun$compileSources$2$adapted(AnalyzingCompiler.scala:335)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:376)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:383)
[error] sbt.internal.inc.AnalyzingCompiler$.compileSources(AnalyzingCompiler.scala:335)
[error] sbt.internal.inc.ZincComponentCompiler.$anonfun$compileAndInstall$3(ZincComponentCompiler.scala:271)
[error] sbt.internal.inc.ZincComponentCompiler.$anonfun$compileAndInstall$3$adapted(ZincComponentCompiler.scala:256)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:376)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:383)
[error] sbt.internal.inc.ZincComponentCompiler.$anonfun$compileAndInstall$2(ZincComponentCompiler.scala:256)
[error] scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] sbt.internal.util.BufferedLogger.bufferQuietly(BufferedLogger.scala:106)
[error] sbt.internal.inc.ZincComponentCompiler.$anonfun$compileAndInstall$1(ZincComponentCompiler.scala:256)
[error] sbt.internal.inc.ZincComponentCompiler.$anonfun$compileAndInstall$1$adapted(ZincComponentCompiler.scala:253)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:376)
[error] sbt.io.IO$.withTemporaryDirectory(IO.scala:383)
[error] sbt.internal.inc.ZincComponentCompiler.compileAndInstall(ZincComponentCompiler.scala:253)
[error] sbt.internal.inc.ZincComponentCompiler.$anonfun$compiledBridgeJar$1(ZincComponentCompiler.scala:219)
[error] sbt.internal.inc.IfMissing$Define.run(IfMissing.scala:19)
[error] sbt.internal.inc.ZincComponentManager.createAndCache$1(ZincComponentManager.scala:46)
[error] sbt.internal.inc.ZincComponentManager.$anonfun$files$3(ZincComponentManager.scala:57)
[error] sbt.internal.inc.ZincComponentManager.getOrElse$1(ZincComponentManager.scala:38)
[error] sbt.internal.inc.ZincComponentManager.$anonfun$files$2(ZincComponentManager.scala:57)
[error] sbt.internal.inc.ZincComponentManager$$anon$1.call(ZincComponentManager.scala:87)
[error] xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
[error] xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
[error] xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[error] xsbt.boot.Using$.withResource(Using.scala:10)
[error] xsbt.boot.Using$.apply(Using.scala:9)
[error] xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
[error] xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
[error] xsbt.boot.Locks$.apply0(Locks.scala:31)
[error] xsbt.boot.Locks$.apply(Locks.scala:28)
[error] sbt.internal.inc.ZincComponentManager.lock(ZincComponentManager.scala:87)
[error] sbt.internal.inc.ZincComponentManager.$anonfun$lockSecondaryCache$1(ZincComponentManager.scala:84)
[... 196 lines cut ...]
[error]         at sbt.internal.inc.IncrementalCompile$.apply(Compile.scala:61)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.compileInternal(IncrementalCompilerImpl.scala:309)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileIncrementally$1(IncrementalCompilerImpl.scala:267)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.handleCompilationError(IncrementalCompilerImpl.scala:158)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.compileIncrementally(IncrementalCompilerImpl.scala:237)
[error]         at sbt.internal.inc.IncrementalCompilerImpl.compile(IncrementalCompilerImpl.scala:68)
[error]         at sbt.Defaults$.compileIncrementalTaskImpl(Defaults.scala:1429)
[error]         at sbt.Defaults$.$anonfun$compileIncrementalTask$1(Defaults.scala:1403)
[error]         at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error]         at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
[error]         at sbt.std.Transform$$anon$4.work(System.scala:66)
[error]         at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error]         at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error]         at sbt.Execute.work(Execute.scala:271)
[error]         at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error]         at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error]         at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[error]         at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
[error]         at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
[error]         at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
[error]         at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
[error]         at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
[error]         at java.base/java.lang.Thread.run(Unknown Source)
[error] (Compile / compileIncremental) java.lang.reflect.InvocationTargetException
[error] Total time: 19 s, completed Jul 31, 2018, 4:07:46 AM

ucbjrl, 2018-07-31:
Could you provide some more details about your environment:
 - OS
 - sbt version
 - java version

eahmedsalman, 2018-08-01:
windows 10 
newest version of both sbt and java

eahmedsalman, 2018-08-01:
i switched to linux and it worked fine but i donot know why its not working on windows

jackkoenig, 2018-08-02:
By "newest version" of java, do you mean Java 10? Scala and SBT have historically been spotty with Java 10. Can you try Java 8?