ucb-bar/chisel-tutorial #140: error sbt run

ipere, 2018-10-24:
Hi, 
I am on  MAC OS, I did as indicated in the README of chisel3
`brew install sbt verilator`
When I wanted to try this tutorial, and that I do sbt run I have : 
```
[info] Loading project definition from /Users/ipere/chisel-tutorial/project
[info] Updating ProjectRef(uri("file:/Users/ipere/chisel-tutorial/project/"), "chisel-tutorial-build")...
Waiting for lock on /Users/ipere/.ivy2/.sbt.ivy.lock to be available...
```

I tried a second time, I have the same thing and it stays frozen for at least 10 minutes, I stopped it then. 

Thanks in advance

closed 2018-10-24 by ipere

jackkoenig, 2018-10-24:
Did you figure out the issue?

ipere, 2018-10-25:
Yes, I tried it twice (the first one was still trying to access in the background) and it seems that another process was using it. Just killed all processes. 
Thanks