ucb-bar/chisel-tutorial #141: Error on generating Verilog

beatsnbytes, 2018-11-11:
Hi all, 
I get the following error when generating Verilog with the command
"./run-examples GCD --backend-name verilator"

/home/beatsnbytes/Documents/Git_Projects/to_experiment/lowrisc/verilator/include/verilated.cpp: In function ‘void vl_hash(bool, std::string)’:
/home/beatsnbytes/Documents/Git_Projects/to_experiment/lowrisc/verilator/include/verilated.cpp:1149:37: error: no matching function for call to ‘std::basic_ifstream<char>::basic_ifstream(std::string&)’
     std::ifstream infile(config_file);
                                     ^
/home/beatsnbytes/Documents/Git_Projects/to_experiment/lowrisc/verilator/include/verilated.cpp:1149:37: note: candidates are:
In file included from /home/beatsnbytes/Documents/Git_Projects/to_experiment/lowrisc/verilator/include/verilated.cpp:35:0:
/usr/include/c++/4.8/fstream:467:7: note: std::basic_ifstream<_CharT, _Traits>::basic_ifstream(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]
       basic_ifstream(const char* __s, ios_base::openmode __mode = ios_base::in)
       ^
/usr/include/c++/4.8/fstream:467:7: note:   no known conversion for argument 1 from ‘std::string {aka std::basic_string<char>}’ to ‘const char*’
/usr/include/c++/4.8/fstream:453:7: note: std::basic_ifstream<_CharT, _Traits>::basic_ifstream() [with _CharT = char; _Traits = std::char_traits<char>]
       basic_ifstream() : __istream_type(), _M_filebuf()
       ^
/usr/include/c++/4.8/fstream:453:7: note:   candidate expects 0 arguments, 1 provided
/usr/include/c++/4.8/fstream:427:11: note: std::basic_ifstream<char>::basic_ifstream(const std::basic_ifstream<char>&)
     class basic_ifstream : public basic_istream<_CharT, _Traits>
           ^
/usr/include/c++/4.8/fstream:427:11: note:   no known conversion for argument 1 from ‘std::string {aka std::basic_string<char>}’ to ‘const std::basic_ifstream<char>&’
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-undefined-bool-conversion" [enabled by default]
make: *** [verilated.o] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory `/home/beatsnbytes/chisel-workspace/chisel-tutorial/test_run_dir/examples/GCD'
================================================================================
Errors: 1: in the following tutorials
Tutorial GCD: throwable assertion failed: 
================================================================================

Is there a mismatch of Chisel3 with Verilator. My installed version is 

Verilator 3.922 2018-03-17 rev UNKNOWN_REV

jackkoenig, 2018-11-12:
This is weird--all of those errors are in Verilator itself, I don't see any mention of a file from Chisel or FIRRTL. Did you build Verilator yourself? If so can you try running its tests?

ucbjrl, 2018-11-12:
> 
> This is weird--all of those errors are in Verilator itself, I don't see any mention of a file from Chisel or FIRRTL. Did you build Verilator yourself? If so can you try running its tests?

This typically means Verilator was build with a different compiler than the one found on the default path.