ucb-bar/chisel-tutorial #143: can I generate verilog without sbt ?

balanx, 2019-02-04:
could someone help me to solve the error ? thanks first.

I don't like sbt very much, and try to transform hello.scala to hello.v as below:

1. download 'chisel3_2.12-3.1.6.jar' from https://mvnrepository.com/artifact/edu.berkeley.cs/chisel3
2. download scala-2.12.8
3. add java/bin, scala/bin to PATH environment
4. edit hello.scala

`package intro

import chisel3._
class HelloWorld extends Module {
  val io = IO(new Bundle{})
  printf("hello world\n")
}

object HelloWorld extends App {
  chisel3.Driver.execute(args, () => new HelloWorld)
}`

5. run shell command
`$ scalac -classpath lib/chisel3_2.12-3.1.6.jar -d classes hello.scala`

I got error.
`hello.scala:11: error: Symbol 'type firrtl.util.BackendCompilationUtilities' is missing from the classpath.
This symbol is required by 'trait chisel3.BackendCompilationUtilities'.
Make sure that type BackendCompilationUtilities is in your classpath and check for conflicting dependencies with `-Ylog-classpath`.
A full rebuild may help if 'BackendCompilationUtilities.class' was compiled against an incompatible version of firrtl.util.
  chisel3.Driver.execute(args, () => new HelloWorld)
  ^
one error found
`

6. then I download firrtl_2.12-1.1.6.jar from mvnrepository again, and add to -classpath
`$ scalac -classpath "lib/chisel3_2.12-3.1.6.jar;lib/firrtl_2.12-1.1.6.jar" -d classes hello.scala`

it's OK    :)

7. next, I run scala 
`$ scala -classpath "classes;lib/chisel3_2.12-3.1.6.jar;lib/firrtl_2.12-1.1.6.jar;lib/scopt_2.12-3.7.1.jar" intro.HelloWorld`

I got error again,
`[info] [0.006] Elaborating design...
[info] [1.831] Done elaborating.
java.lang.ClassNotFoundException: org.json4s.TypeHints
        at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        at chisel3.Driver$.execute(Driver.scala:160)
        at chisel3.Driver$.execute(Driver.scala:202)
        at intro.HelloWorld$.delayedEndpoint$intro$HelloWorld$1(hello.scala:11)
        at intro.HelloWorld$delayedInit$body.apply(hello.scala:10)
        at scala.Function0.apply$mcV$sp(Function0.scala:34)
        at scala.Function0.apply$mcV$sp$(Function0.scala:34)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
        at scala.App.$anonfun$main$1$adapted(App.scala:76)
        at scala.collection.immutable.List.foreach(List.scala:389)
        at scala.App.main(App.scala:76)
        at scala.App.main$(App.scala:74)
        at intro.HelloWorld$.main(hello.scala:10)
        at intro.HelloWorld.main(hello.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at scala.reflect.internal.util.ScalaClassLoader.$anonfun$run$2(ScalaClassLoader.scala:99)
        at scala.reflect.internal.util.ScalaClassLoader.asContext(ScalaClassLoader.scala:34)
        at scala.reflect.internal.util.ScalaClassLoader.asContext$(ScalaClassLoader.scala:30)
        at scala.reflect.internal.util.ScalaClassLoader$URLClassLoader.asContext(ScalaClassLoader.scala:125)
        at scala.reflect.internal.util.ScalaClassLoader.run(ScalaClassLoader.scala:99)
        at scala.reflect.internal.util.ScalaClassLoader.run$(ScalaClassLoader.scala:91)
        at scala.reflect.internal.util.ScalaClassLoader$URLClassLoader.run(ScalaClassLoader.scala:125)
        at scala.tools.nsc.CommonRunner.run(ObjectRunner.scala:22)
`

edwardcwang, 2019-02-04:
Unfortunately, pulling the chisel3 JAR from Maven is not enough since it doesn't contain all the dependencies of Chisel. (Those JARs are referred to as 'far jars'.)

Thankfully there is another option to enable single-file Chisel designs without sbt, and you can check out the demo here: https://github.com/edwardcwang/chisel-single-file

By default though we still recommend the full Chisel template as it provides a standardized structure for building and testing your designs.

balanx, 2019-02-06:
amm has downloaded dependency jar file from maven. I copy all .jar to ./lib and scalac again.
`
scalac -classpath "\
./lib/antlr-runtime-3.5.2-sources.jar;\
./lib/antlr-runtime-3.5.2.jar;\
./lib/antlr4-4.7.1-sources.jar;\
./lib/antlr4-4.7.1.jar;\
./lib/antlr4-runtime-4.7.1-sources.jar;\
./lib/antlr4-runtime-4.7.1.jar;\
./lib/chisel3_2.11-3.1.6-sources.jar;\
./lib/chisel3_2.11-3.1.6.jar;\
./lib/firrtl_2.11-1.1.6-sources.jar;\
./lib/firrtl_2.11-1.1.6.jar;\
./lib/icu4j-58.2-sources.jar;\
./lib/icu4j-58.2.jar;\
./lib/javax.json-1.0.4-sources.jar;\
./lib/javax.json-1.0.4.jar;\
./lib/joda-convert-1.2-sources.jar;\
./lib/joda-convert-1.2.jar;\
./lib/joda-time-2.9.4-sources.jar;\
./lib/joda-time-2.9.4.jar;\
./lib/json4s-ast_2.11-3.5.3-sources.jar;\
./lib/json4s-ast_2.11-3.5.3.jar;\
./lib/json4s-core_2.11-3.5.3-sources.jar;\
./lib/json4s-core_2.11-3.5.3.jar;\
./lib/json4s-native_2.11-3.5.3-sources.jar;\
./lib/json4s-native_2.11-3.5.3.jar;\
./lib/json4s-scalap_2.11-3.5.3-sources.jar;\
./lib/json4s-scalap_2.11-3.5.3.jar;\
./lib/logback-classic-1.2.3-sources.jar;\
./lib/logback-classic-1.2.3.jar;\
./lib/logback-core-1.2.3-sources.jar;\
./lib/logback-core-1.2.3.jar;\
./lib/moultingyaml_2.11-0.4.0-sources.jar;\
./lib/moultingyaml_2.11-0.4.0.jar;\
./lib/nscala-time_2.11-2.14.0-sources.jar;\
./lib/nscala-time_2.11-2.14.0.jar;\
./lib/org.abego.treelayout.core-1.0.3-sources.jar;\
./lib/org.abego.treelayout.core-1.0.3.jar;\
./lib/paranamer-2.8-sources.jar;\
./lib/paranamer-2.8.jar;\
./lib/scala-logging_2.11-3.7.2-sources.jar;\
./lib/scala-logging_2.11-3.7.2.jar;\
./lib/slf4j-api-1.7.25-sources.jar;\
./lib/slf4j-api-1.7.25.jar;\
./lib/snakeyaml-1.17-sources.jar;\
./lib/snakeyaml-1.17.jar;\
./lib/ST4-4.0.8-sources.jar;\
./lib/ST4-4.0.8.jar;\
" -d classes src/intro/GCD.scala
`
I got the same eror again  :(  :(

edwardcwang, 2019-02-06:
Unfortunately for most real development projects, trying to collect all the non-fat JARs manually isn't very feasible, which is why tools like maven, sbt, and ammonite sprung up. See https://stackoverflow.com/questions/3589562/why-maven-what-are-the-benefits for an example of the sorts of issues involved.

grebe, 2019-02-06:
This looks to me like the pesky `-Xsource:2.11` issue. Scala 2.12 changed semantics for anonymous bundles (relied on structural types working in a more permissive way).

In ammonite, the magic incantation is
`interp.configureCompiler(x => x.settings.source.value = scala.tools.nsc.settings.ScalaVersion("2.11.12"))`. In sbt, `scalacOptions += "-Xsource:2.11"` should work.

jackkoenig, 2019-02-06:
We could make a mill `build.sc` for the tutorial. Still a build tool but generally nicer to use than sbt.

balanx, 2019-02-07:
thanks all for your reply.
I used ammonite on  os: windows 10 / cygwin-64 / java 1.8.151 / scala 2.12.8

1)
amm give me an error,

`java.io.IOException: Cannot run program "powershell.exe": CreateProcess error=2, The system cannot find the file specified
`
**Why can't amm use bash in cygwin ?**

2)
then I add powershell to cygwin PATH var, amm run ok. I found

default home directory,
C:\Users\Administrator\.ammonite

all .jar file downloaded in the position,
C:\Users\Administrator\AppData\Local\Coursier\cache

they are different directory !!
**can I put the downloading .jar dir to other place specified by command option ?**
I've tried option '--home', it didn't work.

grebe, 2019-02-07:
I'm not sure what's going on there as I am not a windows user. I think we generally advise windows users to use WSL- sometimes it is necessary to use verilator for simulations, and verilator on windows is tricky unless you use WSL. I suspect it would also resolve the issues you're seeing.

balanx, 2019-02-14:
bad things ! we can't leave sbt.  
I feel Chisel is neither a library nor a new language now.
long long way to go.

chick, 2019-02-14:
I'd suggest starting with [chisel-bootcamp](/freechipsproject/chisel-bootcamp). The examples are newer and it is the primary place to learn chisel. It should get you going without any need to use sbt.

balanx, 2019-02-21:
God blessing !
It works OK when I degrade scala ftom 2.12 to 2.11.

I create a repo on 'https://github.com/balanx/chisel-make'.
GCD runs success with ~28MB jars only.
thanks all