ucb-bar/chisel-tutorial #144: sbt run get error

botaichang, 2019-02-18:
[info] Attempting to fetch org.scala-sbt:compiler-bridge_2.12:1.2.5.
[warn]  [FAILED     ] org.scala-sbt#compiler-bridge_2.12;1.2.5!compiler-bridge_2.12.jar(src): java.lang.NullPointerException (178ms)
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  ::          UNRESOLVED DEPENDENCIES         ::
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  :: org.scala-sbt#util-interface;1.2.2: java.lang.NullPointerException
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[error] The compiler bridge sources org.scala-sbt:compiler-bridge_2.12:1.2.5:compile could not be retrieved.
[error]
[error]         Note: Unresolved dependencies path:
[error]                 org.scala-sbt:util-interface:1.2.2
[error]                   +- org.scala-sbt:compiler-interface:1.2.5
[error]                   +- org.scala-sbt:compiler-bridge_2.12:1.2.5

hattonl, 2019-04-18:
using a faster and stable software to download scientifically

botaichang, 2019-04-19:
> using a faster and stable software to download scientifically

thanks a lot and i have already run it successfully.

closed 2019-04-19 by botaichang