ucb-bar/chisel-tutorial #149: how to run chisel-tutorial in the IntelliJ IDE with Scala and SBT plug-ins?

Ravenwater, 2019-06-16:
I imported the chisel-tutorial into IntelliJ and it recognized the sbt project. Inside the sbt console I can successfully run the builds and tests, but when I try to use the editor, the environment can't resolve the chisel3 import.

Being new to scala, sbt, chisel, and the IntelliJ scala/sbt environment, I need a little help finding the pieces in the environment that are not configured correctly.

schoeberl, 2019-10-17:
> facing the same! please give some solution to this. How can we run this tutorial in the IntelliJ scala environment?

Not super elegant, but you can use the terminal in the IntelliJ editor to type the sbt commands to run your code. So you can still use IntelliJ for editing (that's my default development setup).

There is also a sbt window available within IntelliJ that you can use, it just takes some time to start.

Cheers,
Martin

ShraddhaDevaiya, 2019-10-18:
when I execute the command 'run' in sbt command window to build the definition, it is giving me an error like the following:
[IJ]sbt:chisel-tutorial> run
[warn]
[warn]  Note: Unresolved dependencies path:

[error] stack trace is suppressed; run 'last update' for the full output
[error] (update) sbt.librarymanagement.ResolveException: Error downloading edu.berkeley.cs:chisel-iotesters_2.13:1.2.+
[error]   No version found for [1.2,1.2.max] in
[error]   No version found for [1.2,1.2.max] in

[error]   not found: C:\Users\Pvips_Admin\.ivy2\local\edu.berkeley.cs\chisel-iotesters_2.13
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel-iotesters_2.13/maven-metadata.xml
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel-iotesters_2.13/maven-metadata.xml
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel-iotesters_2.13/maven-metadata.xml
[error] Error downloading edu.berkeley.cs:chisel3_2.13:3.1.+
[error]   No version found for [3.1,3.1.max] in
[error]   No version found for [3.1,3.1.max] in
[error]   not found: C:\Users\Pvips_Admin\.ivy2\local\edu.berkeley.cs\chisel3_2.13
[error]   not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.13/maven-metadata.xml
[error]   not found: https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/chisel3_2.13/maven-metadata.xml
[error]   not found: https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/chisel3_2.13/maven-metadata.xml
[error] Total time: 9 s, completed 18-Oct-2019, 11:21:31 am
[IJ]sbt:chisel-tutorial>
 
so what can I do for this?

ShraddhaDevaiya, 2019-10-31:
Hello,  It is working fine now. there was one JDK  version error, it is supported JDK 11.0 and I had configured JDK 13.0.