ucb-bar/chisel-tutorial #151: why the code of VendingMachineSwitch.scala in the tutorial generate if-else statement instead of  case statement

qgzln, 2019-09-16:
How to generate a case statement for state machine with chisel3

schoeberl, 2019-09-17:
You need to include chisel3.util._ and use a switch. See an example here: https://github.com/schoeberl/chisel-book/blob/master/src/main/scala/SimpleFsm.scala

qgzln, 2019-09-20:
Thank  you for your reply.It is my fault.I didn't make it clear, I mean there is no case in the generated Verilog statement, I usually use the case when writing the state machine with Verilog. So,  chisel3 still can not generate Verilog code with case？

qgzln, 2019-09-20:
> You need to include chisel3.util._ and use a switch. See an example here: https://github.com/schoeberl/chisel-book/blob/master/src/main/scala/SimpleFsm.scala
Thank you for your reply.It is my fault.I didn't make it clear, I mean there is no case in the generated Verilog statement, I usually use the case when writing the state machine with Verilog. So, chisel3 still can not generate Verilog code with case？