ucb-bar/chisel-tutorial #164: sbt run fails with EvalException

DDRDmakar, 2020-06-04:
Hello, I'm a beginner in chisel.  
My system is:  
x86_64 linux Fedora 30  
I installed sbt (0.13.1) and scala (2.10.6)  
my java:
```
[DDRDmakar@localhost chisel-tutorial-3.1.0]$ java -version
openjdk version "1.8.0_252"
OpenJDK Runtime Environment (build 1.8.0_252-b09)
OpenJDK 64-Bit Server VM (build 25.252-b09, mixed mode)
```
I cloned your repo as it shown in readme  
But I get an exception when trying to run "sbt run".
The console log is here:
```
[DDRDmakar@localhost chisel-tutorial-3.1.0]$ sbt run
Getting org.scala-sbt sbt 1.1.1 ...
downloading https://repo1.maven.org/maven2/org/scala-sbt/sbt/1.1.1/sbt-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#sbt;1.1.1!sbt.jar (71ms)
downloading https://repo1.maven.org/maven2/org/scala-lang/scala-library/2.12.4/scala-library-2.12.4.jar ...
        [SUCCESSFUL ] org.scala-lang#scala-library;2.12.4!scala-library.jar (549ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/main_2.12/1.1.1/main_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#main_2.12;1.1.1!main_2.12.jar (212ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/logic_2.12/1.1.1/logic_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#logic_2.12;1.1.1!logic_2.12.jar (65ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/actions_2.12/1.1.1/actions_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#actions_2.12;1.1.1!actions_2.12.jar (78ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/main-settings_2.12/1.1.1/main-settings_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#main-settings_2.12;1.1.1!main-settings_2.12.jar (117ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/run_2.12/1.1.1/run_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#run_2.12;1.1.1!run_2.12.jar (63ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/command_2.12/1.1.1/command_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#command_2.12;1.1.1!command_2.12.jar (80ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/collections_2.12/1.1.1/collections_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#collections_2.12;1.1.1!collections_2.12.jar (87ms)
downloading https://repo1.maven.org/maven2/org/scala-lang/modules/scala-xml_2.12/1.0.6/scala-xml_2.12-1.0.6.jar ...
        [SUCCESSFUL ] org.scala-lang.modules#scala-xml_2.12;1.0.6!scala-xml_2.12.jar(bundle) (108ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/launcher-interface/1.0.2/launcher-interface-1.0.2.jar ...
        [SUCCESSFUL ] org.scala-sbt#launcher-interface;1.0.2!launcher-interface.jar (60ms)
downloading https://repo1.maven.org/maven2/org/apache/logging/log4j/log4j-api/2.8.1/log4j-api-2.8.1.jar ...
        [SUCCESSFUL ] org.apache.logging.log4j#log4j-api;2.8.1!log4j-api.jar (78ms)
downloading https://repo1.maven.org/maven2/org/apache/logging/log4j/log4j-core/2.8.1/log4j-core-2.8.1.jar ...
        [SUCCESSFUL ] org.apache.logging.log4j#log4j-core;2.8.1!log4j-core.jar (185ms)
downloading https://repo1.maven.org/maven2/org/apache/logging/log4j/log4j-slf4j-impl/2.8.1/log4j-slf4j-impl-2.8.1.jar ...
        [SUCCESSFUL ] org.apache.logging.log4j#log4j-slf4j-impl;2.8.1!log4j-slf4j-impl.jar (60ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/io_2.12/1.1.4/io_2.12-1.1.4.jar ...
        [SUCCESSFUL ] org.scala-sbt#io_2.12;1.1.4!io_2.12.jar (204ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/util-logging_2.12/1.1.2/util-logging_2.12-1.1.2.jar ...
        [SUCCESSFUL ] org.scala-sbt#util-logging_2.12;1.1.2!util-logging_2.12.jar (82ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/librarymanagement-core_2.12/1.1.3/librarymanagement-core_2.12-1.1.3.jar ...
        [SUCCESSFUL ] org.scala-sbt#librarymanagement-core_2.12;1.1.3!librarymanagement-core_2.12.jar (126ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/librarymanagement-ivy_2.12/1.1.3/librarymanagement-ivy_2.12-1.1.3.jar ...
        [SUCCESSFUL ] org.scala-sbt#librarymanagement-ivy_2.12;1.1.3!librarymanagement-ivy_2.12.jar (114ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/compiler-interface/1.1.1/compiler-interface-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#compiler-interface;1.1.1!compiler-interface.jar (65ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/zinc-compile_2.12/1.1.1/zinc-compile_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#zinc-compile_2.12;1.1.1!zinc-compile_2.12.jar (58ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/util-relation_2.12/1.1.2/util-relation_2.12-1.1.2.jar ...
        [SUCCESSFUL ] org.scala-sbt#util-relation_2.12;1.1.2!util-relation_2.12.jar (61ms)
downloading https://repo1.maven.org/maven2/com/eed3si9n/sjson-new-scalajson_2.12/0.8.2/sjson-new-scalajson_2.12-0.8.2.jar ...
        [SUCCESSFUL ] com.eed3si9n#sjson-new-scalajson_2.12;0.8.2!sjson-new-scalajson_2.12.jar (61ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/util-position_2.12/1.1.2/util-position_2.12-1.1.2.jar ...
        [SUCCESSFUL ] org.scala-sbt#util-position_2.12;1.1.2!util-position_2.12.jar (67ms)
downloading https://repo1.maven.org/maven2/com/eed3si9n/sjson-new-core_2.12/0.8.2/sjson-new-core_2.12-0.8.2.jar ...
        [SUCCESSFUL ] com.eed3si9n#sjson-new-core_2.12;0.8.2!sjson-new-core_2.12.jar (133ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/completion_2.12/1.1.1/completion_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#completion_2.12;1.1.1!completion_2.12.jar (81ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/task-system_2.12/1.1.1/task-system_2.12-1.1.1.jar ...
        [SUCCESSFUL ] org.scala-sbt#task-system_2.12;1.1.1!task-system_2.12.jar (68ms)
downloading https://repo1.maven.org/maven2/org/scala-sbt/tasks_2.12/1.1.1/tasks_2.12-1.1.1.jar ...
[... 162 lines cut ...]
        at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:57)
        at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:57)
        at xsbt.boot.Launch$.withContextLoader(Launch.scala:77)
        at xsbt.boot.Launch$.run(Launch.scala:57)
        at xsbt.boot.Launch$$anonfun$explicit$1.apply(Launch.scala:45)
        at xsbt.boot.Launch$$anonfun$explicit$1.apply(Launch.scala:45)
        at xsbt.boot.Launch$.launch(Launch.scala:65)
        at xsbt.boot.Launch$.explicit(Launch.scala:45)
        at xsbt.boot.Launch$.initialized(Launch.scala:41)
        at xsbt.boot.Launch$.parsed(Launch.scala:34)
        at xsbt.boot.Launch$.configured(Launch.scala:24)
        at xsbt.boot.Launch$.apply(Launch.scala:19)
        at xsbt.boot.Launch$.apply(Launch.scala:16)
        at xsbt.boot.Boot$.runImpl(Boot.scala:32)
        at xsbt.boot.Boot$.run(Boot.scala:26)
        at xsbt.boot.Boot$.main(Boot.scala:21)
        at xsbt.boot.Boot.main(Boot.scala)
[error] sbt.compiler.EvalException: Type error in expression
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? i
[warn] Ignoring load failure: no project loaded.
[error] Not a valid command: run
[error] run
[error]    ^
```

ucbjrl, 2020-06-04:
This is quite an old version of chisel-tutorial (excuse us for not tagging a more recent release). Please try cloning the current head of the release branch:
```
git clone https://github.com/ucb-bar/chisel-tutorial.git
cd chisel-tutorial
git checkout release
git pull
```

ucbjrl, 2020-06-04:
I've just bumped the current release to match the tip of the release branch.

**NOTE**: we expect you to build chisel-tutorial from source, by cloning the GitHub repository and checking out the release branch.

The appropriate versions of `sbt` and `Scala` will be selected by the build.sbt in the release branch commits.

DDRDmakar, 2020-06-04:
Thanks for the quick support!   
I cloned branch "origin/release" into empty directory and tried "sbt run" again, but get the same error.  
Am I right, that it's enough to have this installed to build chisel tutorial?
- java  (jdk and jre)
- sbt  
- scala  
Or there are another dependences?

ucbjrl, 2020-06-04:
`java` and `sbt` are the only dependencies. `sbt` should download any required dependencies (including the correct version of `Scala`). The errors look like the `sbt` version is incorrect. Could you please try running the following in the chisel-tutorial folder:
```
sbt about
```
and past the results in this issue.

DDRDmakar, 2020-06-04:
```
        ...
        ...
        at xsbt.boot.Boot$.run(Boot.scala:26)
        at xsbt.boot.Boot$.main(Boot.scala:21)
        at xsbt.boot.Boot.main(Boot.scala)
[error] sbt.compiler.EvalException: Type error in expression
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? i
[warn] Ignoring load failure: no project loaded.
[info] This is sbt 1.3.10
[info] No project is currently loaded
[info] sbt, sbt plugins, and build definitions are using Scala 2.10.6
[DDRDmakar@localhost chisel-tutorial]$
```

ucbjrl, 2020-06-04:
There's something fundamentally wrong with your `sbt`. sbt 1.3.10 requires Scala 2.12. You should be seeing something like:
```
[info] Loading project definition from .../chisel-tutorial/project
[info] Loading settings for project chisel-tutorial from build.sbt ...
[info] Set current project to chisel-tutorial (in build file:.../chisel-tutorial/)
[info] This is sbt 1.3.10
[info] The current project is ProjectRef(uri("file:.../chisel-tutorial/"), "chisel-tutorial") 3.3.0
[info] The current project is built against Scala 2.12.10
[info] Available Plugins
[info]  - com.typesafe.sbt.SbtPgp
[info]  - sbt.ScriptedPlugin
[info]  - sbt.plugins.CorePlugin
[info]  - sbt.plugins.Giter8TemplatePlugin
[info]  - sbt.plugins.IvyPlugin
[info]  - sbt.plugins.JUnitXmlReportPlugin
[info]  - sbt.plugins.JvmPlugin
[info]  - sbt.plugins.SbtPlugin
[info]  - sbt.plugins.SemanticdbPlugin
[info] sbt, sbt plugins, and build definitions are using Scala 2.12.10

```

ucbjrl, 2020-06-04:
Could you try installing a version of `sbt` greater than 0.13?

DDRDmakar, 2020-06-05:
Yes, it's something wrong with sbt from fedora repositories.  
I tried another machine with antiX linux and everything works perfect there!  
"sbt about" on antiX gives exactly the same results as in your comment.  
So, it's better to use official repository from SBT website:  
[https://www.scala-sbt.org/download.html]()  
YaY

ucbjrl, 2020-06-05:
Good to hear.

`sbt` 0.13.1 dates from Dec 10, 2013. This is incredible. Fedora 30, whose release was announced in [April, 2019](https://fedoramagazine.org/announcing-fedora-30/), includes [sbt 0.13.1](https://fedora.pkgs.org/30/fedora-x86_64/sbt-0.13.1-9.fc28.1.noarch.rpm.html) as their default (and only) version of `sbt`.

closed 2020-06-05 by ucbjrl