ucb-bar/chisel-tutorial #165: I hope to study chisel but i can't start. please help me.

dldldlfma, 2020-06-07:
Hello! 

## I really hope to study CHISEL.

But it's too hard to me....

Today i tried to this tutorial.

and i couldn't pass at 

```
sbt:chisel-tutorial>test:runMain problems.Launchar Mux4
```

I don't know what is the problem.

---

So, I think i need more background knowledge to start this tutorial. 

### Based on my level, Could i ask you about the road-map to start CHISEL?

i wrote my level at below.
- I use python, C/C++, Verilog
- I don't have any experience about JAVA and Scala. 
- I can use Linux OS like Ubuntu or CentOS

If i take your help, i will try to make a tutorial for beginner who is like me.

I look forward to your reply. 

Thank you and have a nice day!

---

azidar, 2020-07-31:
Have you seen the chisel bootcamp? https://github.com/freechipsproject/chisel-bootcamp

Its a bit easier to start out with.

edwardcwang, 2020-08-01:
I highly recommend the Chisel bootcamp for an interactive, hands-on, and practical introduction to Chisel.

If you prefer a textbook approach, there is an excellent textbook by @schoeberl : https://www.imm.dtu.dk/~masca/chisel-book.html