ucb-bar/chisel-tutorial #166: not found: type SwitchContext

YongkangLi, 2020-09-20:
When I was doing the Mux4, I wanted to use switch-is. But an error occurred during compilation. I know that the referential solution uses two Mux2. I just want to figure out what is wrong doing so. PS: the error message is as follows.

[error] .../chisel-tutorial/src/main/scala/problems/Mux4.scala:38:19: not found: type SwitchContext

[error]   switch (io.sel) {
[error]                   ^
[error] one error found

```scala
class Mux4 extends Module {
  val io = IO(new Bundle {
    val in0 = Input(UInt(1.W))
    val in1 = Input(UInt(1.W))
    val in2 = Input(UInt(1.W))
    val in3 = Input(UInt(1.W))
    val sel = Input(UInt(2.W))
    val out = Output(UInt(1.W))
  })

  //Implement below ----------
  switch (io.sel) {
    is (0.U) {
      io.out := io.in0
    }
    is (1.U) {
      io.out := io.in1
    }
    is (2.U) {
      io.out := io.in2
    }
    is (3.U) {
      io.out := io.in3
    }
  }

  // make the compile process happy, needs to be substituted by the output of the Mux
  //io.out := 1.U
  //Implement above ----------
}
```

jackkoenig, 2020-09-22:
There's a minor bug in the implementation of `switch` that assumes you're doing a wildcard import `import chisel3.util._`. You should be able to workaround by using the wildcard, or you can import `SwitchContext directly, `import chisel3.util.SwitchContext`. This is dumb and an easy fix though so I'll have a PR shortly.

YongkangLi, 2020-09-22:
Thanks for explanation and help. It seems that this was a stupid question. Oops!

closed 2020-09-22 by YongkangLi

jackkoenig, 2020-09-22:
> It seems that this was a stupid question. Oops!

Not at all, `SwitchContext` is supposed to be hidden from you, this is our mistake 😉